ros-controls/ros2_control #964: Add test for the spawner when controller_type parameter is used

destogl, 2023-03-03:
## Background

Check the PR #957 where this functionality was not working.
We should have a test for it in the [controller_manager/test/test_spawner_unspawner.cpp](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/test/test_spawner_unspawner.cpp).
The file needs to be checked if this case is tested and if not add test for it. You can also check if other paramters from spawner are already tested and if not try to test them.

Have a fun!

## Instructions

Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/galactic/Tutorials/Workspace/Creating-A-Workspace.html), for Step3 use "Download Source Code" section with [these instructions](https://ros-controls.github.io/control.ros.org/getting_started.html#compiling).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://ros-controls.github.io/control.ros.org/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/galactic/Tutorials.html)
* [ROS Answers](https://answers.ros.org/questions/)

**Good luck with your first issue!**

Nils-ChristianIseke, 2023-11-03:
See #1152. As far as i get it the test for the type parameter is already implemented.
I am planning to write the tests for all parameters. So Feedback on my PR would be quiet helpful. @destogl

Vishwanath-Karne, 2025-03-02:
It looks like this issue was addressed some time back. Could you provide an update on its current status? Thank you!
@destogl 

Thank you for the detailed explanation! As a 1st contributor, most of my doubts are now cleared. I believe this issue was closed long ago could you confirm? Appreciate your time!

christophfroehlich, 2025-03-02:
@saikishor you may have the best overview, is this still needed? what about #1160?

saikishor, 2025-03-02:
@Vishwanath-Karne @christophfroehlich I'm sure this part is already tested. This part of the code should test it already : https://github.com/ros-controls/ros2_control/blob/3aa38b02283aa6bd04e288919de2724871f599af/controller_manager/test/test_spawner_unspawner.cpp#L332-L397

I believe we can close it

closed 2025-03-02 by christophfroehlich