ros-controls/ros2_control #980: Parameter loading for Humble has changed in the last few months

AndyZe, 2023-03-30:
**Describe the bug**
A debian which was working a few months ago for me no longer works:  [https://github.com/pal-robotics/tiago_robot/blob/humble-devel/tiago_controller_configuration/launch/head_controller.launch.py#L25-L27](https://github.com/pal-robotics/tiago_robot/blob/humble-devel/tiago_controller_configuration/launch/head_controller.launch.py#L25-L27)

There is an error like this for all controllers:

```
tiago_behavior_demos-sim-1  | [spawner-9] [INFO] [1680183611.944682539] [spawner_arm_controller]: Loaded arm_controller
tiago_behavior_demos-sim-1  | [spawner-9] Traceback (most recent call last):
tiago_behavior_demos-sim-1  | [spawner-9]   File "/opt/ros/humble/lib/controller_manager/spawner", line 33, in <module>
tiago_behavior_demos-sim-1  | [spawner-9]     sys.exit(load_entry_point('controller-manager==2.24.1', 'console_scripts', 'spawner')())
tiago_behavior_demos-sim-1  | [spawner-9]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 212, in main
tiago_behavior_demos-sim-1  | [spawner-9]     load_parameter_file(node=node, node_name=prefixed_controller_name, parameter_file=param_file,
tiago_behavior_demos-sim-1  | [spawner-9]   File "/opt/ros/humble/lib/python3.10/site-packages/ros2param/api/__init__.py", line 149, in load_parameter_file
tiago_behavior_demos-sim-1  | [spawner-9]     raise RuntimeError('Param file does not contain parameters for {}, '
tiago_behavior_demos-sim-1  | [spawner-9] RuntimeError: Param file does not contain parameters for arm_controller,  only for nodes: dict_keys(['/arm_controller'])
```

I've just found out, the package maintainer fixed it here:   https://github.com/pal-robotics/tiago_robot/issues/20

So I guess we can close this issue soon, but it might be good to document this ABI change for others.

destogl, 2023-04-16:
I don't understand the issue here. Is it really in the library or on the `tiago_robot` repository? We have changed the mechanism how parameters are handled and loaded in the spawner, but that was not braking change in any file that had proper definition of parameters. From the output it seems to me that some parameters were missing earlier too, but this was just not visible.

bmagyar, 2023-04-26:
Closing now due to lack of activity. Please reopen if there's a clearer picture of what's wrong.

closed 2023-04-26 by bmagyar