ros-controls/ros2_control #988: GenericSystem - all storage vectors initialize to size of joints

nathandunk, 2023-04-13:
**Describe the bug**

In Humble (although it looks like the same issue is present in galactic), when using `fake_components/GenericSystem` when there are no joints initialized, but there are other ros2_control items (sensor, gpio, etc), the interfaces do not get properly initialized in the `GenericSystem::initialize_storage_vectors()` method. These size the second index of the `commands` and `states` vectors to `info_.joints.size()`, rather than the correct component info, such as `info_.sensors.size()` when sensors are initialized. If there are no joints present, this results in trying to access an index of a vector of 0 size which throws a runtime error.

It looks like the correct fix is already available in the master branch, so it would be useful it this could be provided to Humble as well.

I can go ahead and make the fix and submit a PR if that is more useful. haven't done that before, but would be happy to contribute! I first wanted to make sure I had the right idea of the correct behavior.

**To Reproduce**

The minimal steps to reproduce this is:

- Use the galactic branch of the ros2_control_demos repository, and make the necessary changes to have it compile in humble
  - I think just this is just adding new arguments in `read()` and `write()` methods on hardware interfaces, and
  - Adding `using hardware_interface::CallbackReturn;` where necessary
- Remove the joints from `rrbot_system_with_sensor.ros2_control.xacro`
- Remove everything but `joint_state_broadcaster` from `rrbot_with_sensor_controllers.yaml`
- Remove all nodes except `control_node`, `robot_state_pub_node`, and `joint_state_broadcaster_spawner` from launch files
- Run `ros2 launch ros2_control_demo_bringup rrbot_system_with_sensor.launch.py use_fake_hardware:=true` and receive the following output
 

```
[ros2_control_node-1] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-1]   what():  double hardware_interface::ReadOnlyHandle::get_value() const failed. value_ptr_ is null.
[ros2_control_node-1] Stack trace (most recent call last) in thread 8992:
[ros2_control_node-1] #13   Object "", at 0xffffffffffffffff, in
[ros2_control_node-1] #12   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 81, in __clone3 [0x7f36dbd269ff]
[ros2_control_node-1] #11   Source "./nptl/pthread_create.c", line 442, in start_thread [0x7f36dbc94b42]
[ros2_control_node-1] #10   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f36dc0dc2b2, in
[ros2_control_node-1] #9    Object "/opt/ros/humble/lib/controller_manager/ros2_control_node", at 0x55a2cf88883b, in
[ros2_control_node-1] #8    Object "/opt/ros/humble/lib/libcontroller_manager.so", at 0x7f36dc6b4047, in controller_manager::ControllerManager::update(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-1] #7    Object "/opt/ros/humble/lib/libjoint_state_broadcaster.so", at 0x7f36c4747c33, in
[ros2_control_node-1] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f36dc0ae517, in __cxa_throw
[ros2_control_node-1] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f36dc0ae2b6, in std::terminate()
[ros2_control_node-1] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f36dc0ae24b, in
[ros2_control_node-1] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f36dc0a2bbd, in
[ros2_control_node-1] #2    Source "./stdlib/abort.c", line 79, in abort [0x7f36dbc287f2]
[ros2_control_node-1] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7f36dbc42475]
[ros2_control_node-1] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[ros2_control_node-1]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[ros2_control_node-1]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7f36dbc96a7c]
[ros2_control_node-1] Aborted (Signal sent by tkill() 8966 1000)
```

**Expected behavior**

Program should initialize storage vectors with the respective sizes of the component infos, and therefore not crash.

**Environment:**

OS: Ubuntu 22.04
Version: Humble

Thanks!

destogl, 2023-04-16:
@nathandunk Thanks for bringing this to our attention! Can you please check if #991 solves your issue? I think it should.

nathandunk, 2023-04-18:
Yes, that looks like it solves it! However, it looks like some testing infrastructure is missing to go with the `test_generic_system.cpp` edits that are included in the backport. Thanks!

destogl, 2023-04-20:
@nathandunk yes, I am on it to fix those and backport stuff

closed 2023-04-20 by destogl