ros-controls/ros2_control #995: [Chained Controllers] Chaining of State Interfaces

destogl, 2023-04-20:
The controllers chaining described in [Cascade Control Proposal](https://github.com/ros-controls/roadmap/blob/master/design_drafts/cascade_control.md) is not support only chaining of command interface between controllers.
The functionality is based on the same principles for communication between controllers and hardware, where hardware offers `Command Interfaces` and controllers as loaning them based on the name matching using `Loaned Command Interfaces`.
Since such controllers need additional fields and methods, a new type of base controller class is introduced called `ChainableControllerInterface`.
This interface can be used for both classical controllers and chainable controllers. So if you are uncertain if your controller should be chainable in the future, you can implement this base class and simply export an empty list of reference interfaces and disable switching to chained mode.
Those interfaces are called `Reference Interfaces` in the controllers (still have the same type of `CommandInterfaces`).

`Reference Interfaces` are exported when a `Chainable Controller` is configured and stored in the resource manager. When a controller is to be activated, controller manager check if requested command interfaces are from another chainable controllers and if so, it confirms that the following controller is active.

**NOTE** currently the controllers have to be loaded in order from the beginning toward the end of the chain 

For chaining state interfaces, the same logic can be used:

1. Extend existing `ChainableControllerInterface` with method for exporting `Estimated Interfaces` (type `StateInterface`)
2. Get those exported interfaces when configuring controllers in controller manager ([somewhere here](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/src/controller_manager.cpp#L505-L521)
3. Understand the logic of current controller switching for the chained controllers and extend it to support also chaining of state interfaces – begins here](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/src/controller_manager.cpp#L539) --> basically the only trick is to properly fill the activation and deactivation lists.
4. Dr. Denis recommends to actually write first the tests with the controller chain setup (a figure is also helpful) – because then it will be clear which situations should be covered. [Check the tests here](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/test/test_controllers_chaining_with_controller_manager.cpp).

saikishor, 2023-04-20:
Thank you Denis for the nice description. I will take this and work on it. :)

saikishor, 2023-12-06:
https://github.com/orgs/ros-controls/projects/6/views/9?sliceBy%5Bvalue%5D=Jazzy&pane=issue&itemId=46788689

closed 2024-07-05 by destogl