ros-controls/ros2_control #1016: How to pass additional information to hardware interface

guidout, 2023-05-10:
## Background

Hi,
I have posted a question on [Stack Exchange](https://robotics.stackexchange.com/questions/24738/how-to-exchange-commands-and-states-between-hardware-interfaces-to-generate-a-c) about CAN messages and hardware interfaces but I think the question can be generalized.

My issue (maybe it's already possible but I'm not aware of how) is how to pass additional information to the hardware interface besides commands and states.
In my use case, I have to interface with an existing vehicle CAN bus (so I can't change the messages' definitions) and the hardware interface will send steering and throttle commands to the existing vehicle module.
Each CAN message has more information than just the command and/or the sensor. So I need a way to get that information down to the hardware interface so that it can generate a complete CAN message.
This additional information could come either from a node or from a controller.

Is there an already existing mechanism to accomplish this? 

Thank you in advance

destogl, 2023-05-30:
yes, please check how we have solved this in the ros2_canopen repository. There you can find ideas, the right amount of abstraction you have to do yourself.

@Xi-HHHM is just adding detailed docs about ros2_canopen so [this can also help you to understand the concepts](https://github.com/StoglRobotics-forks/ros2_canopen/blob/cia301_documentation/canopen/sphinx/user-guide/operation/ros2-control-interface.rst) - note that this description is subject to changes – we just want to structure this better.

mateusmenezes95, 2024-02-01:
This is an old issue, but for new people who are looking for a solution, an alternative to ros2_canopen could be the feature proposed in the PR https://github.com/ros-controls/ros2_control/pull/1240

closed 2025-05-20 by github-actions[bot]