ros-controls/ros2_control #1033: How to reconfigure controller

firesurfer, 2023-05-25:
**Is your feature request related to a problem? Please describe.**

I would like to reconfigure controller which already has been active at some point in order to change a parameter which is read in the on_configure method of the controller.

In particular I want to change the `compliance_ref_link` of the the `cartesian_compliance_controller` to another link:

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

firesurfer, 2023-07-17:
As there weren't any answers yet I guess this functionality is not implemented yet?

destogl, 2023-07-18:
Hi @firesurfer,

you have two options here. Use controller lifecycle to again configure the controller, or you can use dynamic reconfigure. `generate_parameter_library` that we use integrated this by default. Then you can update parameters in any state – which might be possible depending on the controller or use-case.

You should be able to set controller state via CLI, either using `set_controller_state` service (not sure if fully integrated yet) or calling `switch_controllers` to deactivate it and then `cleanup` to unconfigure it.

firesurfer, 2024-02-20:
Hi  @destogl  sorry for my really late answer. For some reason I sometimes do not get a notification email from github. 

I currently worked around the issue by simply having multiple controllers for all configurations that I need. I recently revisited that part of the setup and I still think it is a very ugly solution. As it will be unlikely that the cartesian compliance controllers will be changed soon I come back to this issue.

So to recap the initial issue:

The cartesian compliance controller setup their kinematic chains in the `on_configure` method. 
Ideally the workflow would then be: Bring the controller into `unconfigured state` set the parameters configure and activate it again.

As far as I can see there is no way currently of having the controller run `on_configure` again once it has been loaded with the provided cli tools? 
I am note sure how I am supposed to call cleanup via the cli or a ros service?

christophfroehlich, 2024-02-20:
Is it feasible to do that in `on_activate` instead of `on_configure`? Our current best-practice is afaik

- declare parameters in the `on_init`
- parse all parameters and check read-only parameters in `on_configure` manually, if not done already by generate_parameter_library like combination of parameters
- if parameters changed, parse them again in `on_activate` and check dynamic parameters manually, if not done already by generate_parameter_library like combination of parameters
- if desired, do that again in the `update` method, e.g., controller gains.

firesurfer, 2024-02-20:
That is basically what I implemented in: https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/pull/129

Unfortunately this introduces some rather heavy load in the `on_activate` method. Additionally it doesn't look like  @stefanscherzinger won't find anytime soon to review that PR. 

Nevertheless I think it would be a great addition to ros2control if we were able to rerun on_configure in a controller!.

saikishor, 2024-02-20:
> That is basically what I implemented in: [fzi-forschungszentrum-informatik/cartesian_controllers#129](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/pull/129)
> 
> Unfortunately this introduces some rather heavy load in the `on_activate` method. Additionally it doesn't look like @stefanscherzinger will find anytime soon to review that PR.
> 
> Nevertheless I think it would be a great addition to ros2control if we were able to rerun on_configure in a controller!.

Hello!

Right now, the only way to configure a controller is by deactivating the controller and calling again the configure_controller service
https://github.com/ros-controls/ros2_control/blob/c4affe4bbfbb12b947acfc7b268f529724e0aae8/controller_manager/src/controller_manager.cpp#L735-L750

(or)

`deactivate the controller` -> `unload the controller` -> `load the controller` -> `configure the controller`

However, @destogl and @bailaC are working on a new service called [cleanup_controller](https://github.com/ros-controls/ros2_control/pull/1236) or unconfigure_controller, where you can unconfigure the controller and configure it again through services. This will make it upstream soon

firesurfer, 2024-02-20:
That sounds good. 
But in case of cleanup being called will the changed parameters survive? 

This is the problem with unloading the controller in this case and reloading it again because in this case the node is destroyed and on reload populated with the parameters from the `controller.yaml`.

saikishor, 2024-02-20:
> But in case of cleanup being called will the changed parameters survive?

I'm not sure about this part, please try it out and let us know. It would be really helpful

Thank you

firesurfer, 2024-02-20:
So I just tested it and it seems to be successful (apart from the fact that the cartesian controller checks if it was already configured and then returns immediately - but this is another problem)

What I did was:

a) Load one one of the cartesian controllers
b) Changed one of the parameters via ros2 param set
c) Deactivated the controller
d) Ran ` ros2 service call /controller_manager/configure_controller controller_manager_msgs/srv/ConfigureController "{name: cartesian_motion_controller_top}"`
e) Activated the controller again
f) Checked if the changed param is the same  via ros2 param get

saikishor, 2024-02-20:
@firesurfer Thanks good to know about this
So, once you fix that `m_configured` parameter, I think it should work straight out of the box

firesurfer, 2024-06-19:
Would be nice if this call could be integrated in the cli tooling. 
But I close this issue as it is solved!

closed 2024-06-19 by firesurfer

srsidd, 2025-06-06:
@firesurfer, I followed your comment thread from [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/127), and tried to [recreate your steps](https://github.com/ros-controls/ros2_control/issues/1033#issuecomment-1954069949) to reload the cartesian_compliance_controller. However, step d  ros2 service call /controller_manager/configure_controller controller_manager_msgs/srv/ConfigureController "name: 'cartesian_compliance_controller'" always returns `False`. Any suggestions on how to get past that?

firesurfer, 2025-06-07:
1. The `{  }` in the message are important from my understanding
2. Is the controller loaded and does it have that name?

> So I just tested it and it seems to be successful (apart from the fact that the cartesian controller checks if it was already configured and then returns immediately - but this is another problem)

We actually do not reconfigure the controller at the moment but have 2 different controllers for different end effectors / compliant frames and load/activate the one we need.

srsidd, 2025-06-18:
1. Looking at the srv definition [here](https://docs.ros.org/en/iron/p/controller_manager_msgs/interfaces/srv/ConfigureController.html), the brackets would not result in the correct behavior I would think, as that would make it into a list instead of a string.
2. The controller was loaded with the right name. 

Understood!