ros-controls/ros2_control #1045: Getting to a flexible components management from URDF in `/robot_description` topic

destogl, 2023-06-05:
This description of issues is related to the following PRs:

- ros-controls/ros2_control_demos#277
- ros-controls/ros2_control#940

To solve this knot, I figured out the following 4 steps that will be done in separate PRs to make them manageable:

1. Rename parameters for setting components in `unconfigured` and `inactive` state after CM is started. This feature does not break anything, and it should be back ported to `humble` as well. See: #
2. Enable loading components from topic if they are not loaded from parameter. Only the first callback will be considered to avoid any complexity with managing potential changes of components in URDF (fixing #940).
3. Enable `not_loaded` components after start. This needs some changes in RM (ABI breaking) that will separate some things better.
4. Enable updates of `/robot_description` and components in URDF. The default behavior will be that if a new component exists it will be added to the system, but none component will be removed if it doesn't exist in URDF. User will be notified about this and they can manually remove the component.

christophfroehlich, 2023-06-22:
Ad #940: Can it be that it was forgotten to update the documentation 
https://github.com/ros-controls/ros2_control/blob/db34dfaf2d79fc8afcd460d9ae16f02c43265a2a/controller_manager/doc/userdoc.rst?plain=1#L58-L60
according to this deprecation warning?
https://github.com/ros-controls/ros2_control/blob/db34dfaf2d79fc8afcd460d9ae16f02c43265a2a/controller_manager/src/controller_manager.cpp#L162-L166

destogl, 2023-07-18:
Yes, you are correct, docs has to be updated.