ros-controls/ros2_control #1073: Good practice to handle multiple controller_managers

LucasLabarussiat, 2023-07-06:
Hey everyone,

First time opening an issue, please bear with me if I forget some details in my question. I tried finding an answer to this question online, spent some time looking to the doc and the issues, didn't find what I was looking for. I might have missed it though, as it seems like a basic question.

Current setup:
- Ubuntu 22.04 (not a RT Kernel though)
- ROS2 Humble

Hardware:
- An Universal Robot UR10e
- A Robotiq Hand-E gripper

I am using the Universal_Robots_ROS2_Driver package, and a custom package with a proper hardware_interface implementation for the gripper, and I have a gripper_description package as well.

At first, I was including my gripper.xacro file into ur10e.xacro, meaning that the whole thing was loaded into one _robot_description_ and in a single controller_manager.
When sending a goal to my gripper_controller, it resulted in an error from the [RTDE Client : Pipeline producer overflowed!](https://github.com/UniversalRobots/Universal_Robots_Client_Library#rtdeclient) , which is raised when the robot buffer isn't read.

My guess was that the control node was busy with the gripper and couldn't keep up with the robot at the same time.

So I decided to split these controllers into two different controller_managers. Created a proper launch file for my gripper alone : 

```
from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, OpaqueFunction
from launch.conditions import IfCondition, UnlessCondition
from launch.substitutions import Command, FindExecutable, LaunchConfiguration, PathJoinSubstitution

def launch_setup(context, *args, **kwargs):
    # General arguments
    gripper_runtime_config_package = LaunchConfiguration("gripper_runtime_config_package")
    gripper_controllers_file = LaunchConfiguration("gripper_controllers_file")
    gripper_controller_spawner_timeout = LaunchConfiguration("gripper_controller_spawner_timeout")
    gripper_description_package = LaunchConfiguration("gripper_description_package")
    gripper_description_file = LaunchConfiguration("gripper_description_file")
    gripper_use_fake_hardware = LaunchConfiguration("gripper_use_fake_hardware")

    hande_config_file = PathJoinSubstitution([FindPackageShare(gripper_runtime_config_package), "config", "hande_config.yaml"])
    gripper_initial_joint_controllers = PathJoinSubstitution([FindPackageShare(gripper_runtime_config_package), "config", gripper_controllers_file])
    gripper_update_rate_config_file = PathJoinSubstitution([FindPackageShare(gripper_runtime_config_package), "config", "hande_update_rate.yaml"])
    gripper_ns = LaunchConfiguration("gripper_ns")
    gripper_name = LaunchConfiguration("gripper_name")

    robot_description_content = Command(
        [
            PathJoinSubstitution([FindExecutable(name="xacro")]),
            " ",
            PathJoinSubstitution([FindPackageShare(gripper_description_package), "urdf", gripper_description_file]),
            " ",
            "gripper_name:=",
            gripper_name,
            " ",
            "hande_config_file:=",
            hande_config_file,
            " ",
            # "gripper_use_fake_hardware:=",
            # gripper_use_fake_hardware,
            # " ",
        ]
    )
    robot_description = {"robot_description": robot_description_content}

    control_node = Node(
        package="controller_manager",
        executable="ros2_control_node",
        # name="gripper_controller_manager", # NOT NEEDED WITH NAMESPACES
        parameters=[robot_description, gripper_update_rate_config_file, gripper_initial_joint_controllers],
        output="screen",
[... 128 lines cut ...]
            # push-ros-namespace to set namespace of included nodes
            PushRosNamespace(gripper_ns),
            IncludeLaunchDescription(
                PythonLaunchDescriptionSource(
                    os.path.join(
                        get_package_share_directory("robotiq_hande_hwi"),
                        'launch/sri_hande_control.launch.py'))
            ),
        ],
        condition=IfCondition(with_gripper),
    )
```

I can confirm I can now send goals to both controllers in the same time, and I don't have any "Pipeline producer overflowed" errors. It also feels better to be able to set two different update rates, but I've since then read you could do it in the `<plugin>` tag.

The problem now lies in that I have two different `robot_description` parameters. And I'm not sure how I should link those two URDF files together, for example to see both the robot and the gripper in RViz. At first, I though I'd add a `robot_state_publisher` with my gripper URDF in the dedicated launch file, and I would have 2 Robot Models in RViz. But it can't parse the URDF because it doesn't know where to find the `tool0` joint of my ur10e.
If I try to still use a global xacro file, both controller_managers would load the 2 plugins (which I tried and of course didn't work).

I feel like this must be a pretty common thing to do, and I should have been able to find some best practices about running two controller_managers and still having a whole system description. It will happen again e.g if I put my arm on a mobile base (which should happen sooner or later).
My only guess at the moment would be to create a copy of the gripper xacro file without the ros2_control plugin and include this copy in the ur10e xacro file, which I tried and I'd have to remap the /gripper/joint_states to /joint_states then, which doesn't seem good.

If you have any advice on how to tackle it, I'd be glad to hear it (or you can point me in the right direction if this has already been discussed).

Thanks,
Lucas

saikishor, 2023-07-09:
Hello @LucasLabarussiat,

Looking at your use case, I think what you need is async controllers. If I understand correctly, your gripper controller is blocking the update for other running controllers, so you are facing this issue. However, this wouldn't be the case if it's an async controller as it runs on its own thread, so it wouldn't block other controllers in the controller_manager. Take a look at the following parameter to set it up properly. 

https://github.com/ros-controls/ros2_control/blob/09c6da4a170f56c39f57e22cf59d2d0927eeb6f3/controller_interface/src/controller_interface_base.cpp#L88. 

Best Regards,
Sa
i

LucasLabarussiat, 2023-07-12:
Hello @saikishor ,

Thanks for your quick answer. I haven't been able to try it out until now but will give you an update as soon as possible.
If I understand correctly, I'd have to set this parameter in my controller. To do so, could I add it in my `controllers.yaml` ?

```
/gripper:
  controller_manager:
    ros__parameters:
      joint_state_broadcaster:
        type: joint_state_broadcaster/JointStateBroadcaster

      gripper_controller:
        type: position_controllers/GripperActionController

      # gripper_effort_controller:
      #   type: effort_controllers/GripperActionController

  gripper_controller:
    ros__parameters:
      # Base parameters
      joint: hande_joint_finger
      action_monitor_rate: 20.0 #default 20.0
      allow_stalling: false #default false
      goal_tolerance: 0.01 #default 0.01 ! Isn't taken into account, use "tol" parameter in hande_HWI.hpp
      max_effort: 100.0 #default 0.0 ! Isn't taken into account, use "max_effort" parameter in hande_config.yaml
      stall_timeout: 1.0 #default 1.0
      stall_velocity_threshold: 0.001 #default 0.001

      is_async: true
```

I don't remember seing this parameter when I used `list_parameters`, if I had I would have given it a try for sure.
I have since found a [`non-blocking read`](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#non_blocking_read-default-false) parameter on the UR ROS driver as well. Didn't have a chance to try it either, and I am not sure my set-up is a proper `combined_robot_hw`.

Back to the "good practices" I mentioned, does this mean it would be better to keep a single `controller_manager` then ? (And make sure this `is_async` parameter is properly set on the `gripper_controller` and the `ur_controller`.)
I see there is also an `update_rate` parameter in the `controller_interface_base.cpp`, does this override the `update_rate` of the `controller_manager` ?

Thanks again, with regards,
Lucas

destogl, 2023-07-17:
Hi @LucasLabarussiat,

in your case, is definitely better to have only one controller manager. Usually, one uses multiple ones if you have physical separate hardware that you don't need to synchronize. I am not convinced that you have a problem with the controller, but on the hardware side. So you should use async hardware functionality that came in recently PR #981 (@bmagyar can you please answer this?). If you have blocking calls to your gripper, then you should put it in a separate thread and this should work with one controller manager. This is done automatically using "async" features.

LucasLabarussiat, 2023-07-21:
Hi @destogl,

Thanks for your answer. It might indeed be on the hardware side. I finally had some time to try out the `is_async` parameter of the controllers. I used it on both the UR and the gripper controller and it didn't change much.
I will try to test this on the hardware. I might get back to you soon with another question, as I still have some difficulties understanding the whole control architecture.

It might be interesting to cross-post this issue on the [UR repo](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver) too, what do you think?

LucasLabarussiat, 2023-07-21:
Ok I might need some guidance on this, if this isn't too much to ask.

I added `is_aync` as a parameter in the `hardware` tag of my xacro:
```
<ros2_control name="${name}" type="system">
       <hardware>
              <xacro:if value="${use_fake_hardware}">
                <plugin>mock_components/GenericSystem</plugin>
              </xacro:if>
              <xacro:unless value="${use_fake_hardware}">
                <plugin>robotiq_hardware/HandeHWI</plugin>
                <param name="gripper_ip">${gripper_ip}</param>
                <param name="port">${port}</param>
                <param name="min_speed">${min_speed}</param>
                <param name="max_speed">${max_speed}</param>
                <param name="min_effort">${min_effort}</param>
                <param name="max_effort">${max_effort}</param>
                <param name="min_gap">${min_gap}</param>
                <param name="max_gap">${max_gap}</param>
                <param name="min_gap_counts">${min_gap_counts}</param>
                <param name="is_async">${is_async}</param>
              </xacro:unless>
       </hardware>
```

I set it to true in a yaml file. To make sure this was properly set, I added this in my `hardware_interface.cpp` :
```
is_async_ = info_.hardware_parameters["is_async"];
RCLCPP_INFO(
        rclcpp::get_logger("HandeHWI"), "is_async parameter = %s", is_async_.c_str());
```

I first did it for the gripper, it didn't change my error. Then added it to the UR xacro as well, with the same result.
When you said "this is done automatically", do I only have to add this parameter, or are there more steps ? I didn't seem to find any documentation about it.
Plus I don't know how to check if my hardware interfaces (and controllers) do indeed run on separate threads.

Should I also try to have different update rates for my UR_hardware_interface and gripper_hardware_interface following [these instructions](https://github.com/ros-controls/ros2_control/tree/master/hardware_interface#best-practice-for-having-different-update-rate-for-each-hardware_interface-by-counting-loops).

Would this help to share the files?

closed 2023-08-21 by LucasLabarussiat