ros-controls/ros2_control #1090: Add documentation for async mode of controllers and hardware components

bmagyar, 2023-07-27:
PRs for reference
* https://github.com/ros-controls/ros2_control/pull/981
* https://github.com/ros-controls/ros2_control/pull/932
* https://github.com/ros-controls/ros2_control/pull/927
* https://github.com/ros-controls/roadmap/blob/master/design_drafts/async_controller.md

christophfroehlich, 2025-05-20:
I think this is still not documented well, there is only one page about hardware components but nothing about controllers 
https://control.ros.org/rolling/search.html?q=async&check_keywords=yes&area=default

ShahazadAbdulla, 2026-04-02:
Hi @christophfroehlich 
I was previously assigned to this issue but went quiet and I sincerely apologize for that. I'd like to request one more chance to contribute and complete this myself and through that wanna start contributing more to ros2_control from here on out.

I've since done a thorough review of the existing documentation, the design draft, and the three referenced PR mentioned above. I understand that the async implementation has been rewritten since those PRs, and I like to base the documentation on the current updated codebase.

My plan is to write a dedicated page for async controllers mirroring the structure of the existing hardware async page, covering: what async controllers are and why they exist, how to enable them with configuration, lifecycle behavior, thread management , interaction with the controller manager and diagnostics, and relevant details.

I will have ready the PR shortly.