ros-controls/ros2_control #1095: view_controller_chains does not work with example

christophfroehlich, 2023-08-07:
**Describe the bug**
The output of view_controller_chains does not seem to work with the new example 12 of the demo repository.

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout https://github.com/ros-controls/ros2_control_demos/pull/318
2. Run the example as written in the doc
3. `ros2 control view_controller_chains`

**Expected behavior**
Some chain in the PDF.

**Screenshots**
But there are no links, just nodes:
[controller_diagram.gv.pdf](https://github.com/ros-controls/ros2_control/files/12272927/controller_diagram.gv.pdf)

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Version: Rolling

**Additional context**
none

christophfroehlich, 2023-08-07:
@pac48 could you please have a look on that if you find time?

pac48, 2023-08-11:
I was able to reproduce the error that you mentioned. It seems that the Python library was changed from `graphviz` to `pygraphviz` in this PR  https://github.com/ros-controls/ros2_control/pull/812 When I run with the old `graphviz` way, I get this output.  
[image]
I am not too familiar with the differences between `graphviz` and `pygraphviz`, but it seems there is a difference in usage. 

Is it possible to get a diagram of what the output should look like? The output above doesn't seem right.

christophfroehlich, 2023-08-12:
I haven't seen the changes from your version, sorry. Maybe @sachinkum0009 can have a look?

@saikishor designed the example, imho the graph posted above isn't correct?

bmagyar, 2023-08-12:
I believe the motivation for using a different graphviz python implementation was that the former isn't supported by rosdep

closed 2026-09-02 by christophfroehlich