ros-controls/ros2_control #1103: Ressource manager hangs up on lifecycle changes

christophfroehlich, 2023-08-22:
**Describe the bug**
After changing the lifecycle of a hardware component, no matter if via controller-manager service or from the component's write method returning `hardware_interface::return_type::ERROR`, the controller manager spams the console with something like

```
[ros2_control_node-1] [WARN] [1691260478.510570255] [resource_manager]: (hardware 'RRBotSystemPositionOnly'): 'rrbot_joint1/position' command interface not in available list. This should not happen (hint: multiple cleanup calls).
[ros2_control_node-1] [WARN] [1691260478.510572441] [resource_manager]: (hardware 'RRBotSystemPositionOnly'): 'rrbot_joint2/position' command interface not in available list. This should not happen (hint: multiple cleanup calls).
[ros2_control_node-1] [WARN] [1691260478.510574080] [resource_manager]: (hardware 'RRBotSystemPositionOnly'): 'rrbot_joint1/position' state interface not in available list. This should not happen (hint: multiple cleanup calls).
[ros2_control_node-1] [WARN] [1691260478.510575651] [resource_manager]: (hardware 'RRBotSystemPositionOnly'): 'rrbot_joint2/position' state interface not in available list. This should not happen (hint: multiple cleanup calls).
```

And the component can't be activated again. If this is a result of overflow of the logger, or some failed state of the ressource manager?

**To Reproduce**
Steps to reproduce the behavior:

Follow the instructions of [this demo example](https://github.com/StoglRobotics-forks/ros2_control_demos/pull/7).

**Expected behavior**
A clean deactivation or shutdown of the hw component, being able to reactivate it again.

**Screenshots**

**Environment (please complete the following information):**
 - OS: ubuntu 22.04
 - ros2_control rolling from source

christophfroehlich, 2023-12-28:
We recently also get these error messages with the tests here: https://github.com/ros-controls/ros2_control/actions/runs/7343670255/job/19994517445#step:6:13108

:eyes: @saikishor

saikishor, 2023-12-29:
@christophfroehlich  Thanks for reporting it. You are referring to the below output right?

```  Error: OR] [1703729558.642703802] [mock_generic_system]: Got multiple (2) starting interfaces for joint 'joint1' - this is not supported!
  Error: OR] [1703729558.642715534] [mock_generic_system]: Got multiple (2) starting interfaces for joint 'joint2' - this is not supported!
  Error: OR] [1703729558.642726604] [resource_manager]: Component 'HardwareSystem2dofStandardInterfacesWithDifferentControlModes' did not accept new command resource combination: 
  7:  Start interfaces: 
  7: [
  7:   flange_vacuum/vacuum
  7:   joint1/position
  7:   joint1/velocity
  7:   joint2/acceleration
  7:   joint2/velocity
  7: ]
  7: Stop interfaces: 
  7: [
  7: ]
  ```
  
  I saw a couple of this, when running the following test `7: Test command: /usr/bin/python3.10 "-u" "/opt/ros/rolling/share/ament_cmake_test/cmake/run_test.py" "/home/runner/work/ros2_control/ros2_control/.work/target_ws/build/hardware_interface/test_results/hardware_interface/test_generic_system.gtest.xml" "--package-name" "hardware_interface" "--output-file" "/home/runner/work/ros2_control/ros2_control/.work/target_ws/build/hardware_interface/ament_cmake_gmock/test_generic_system.txt" "--command" "/home/runner/work/ros2_control/ros2_control/.work/target_ws/build/hardware_interface/test_generic_system" "--gtest_output=xml:/home/runner/work/ros2_control/ros2_control/.work/target_ws/build/hardware_interface/test_results/hardware_interface/test_generic_system.gtest.xml"`

christophfroehlich, 2023-12-29:
Oh sorry, the link to the specific line did not work. The `TestSystemHardware` is in unconfigured state, but the rm complains repeatedly
```
  10: [ RUN      ] TestControllerManagerHWManagementSrvs.list_hardware_components
   ...
  10: [WARN] [1703729657.862577856] [resource_manager]: (hardware 'TestSystemHardware'): 'joint2/velocity' command interface not in available list. This should not happen (hint: multiple cleanup calls).
  10: [WARN] [1703729657.862596992] [resource_manager]: (hardware 'TestSystemHardware'): 'joint3/velocity' command interface not in available list. This should not happen (hint: multiple cleanup calls).
  10: [WARN] [1703729657.862601260] [resource_manager]: (hardware 'TestSystemHardware'): 'joint2/max_acceleration' command interface not in available list. This should not happen (hint: multiple cleanup calls).
  10: [WARN] [1703729657.862605888] [resource_manager]: (hardware 'TestSystemHardware'): 'configuration/max_tcp_jerk' command interface not in available list. This should not happen (hint: multiple cleanup calls).
```

saikishor, 2023-12-29:
@christophfroehlich Thank you for the logs. Ok this seems to be getting important now. I'll see if I can find some time to fix it when I'm back from holidays

saikishor, 2024-01-06:
@christophfroehlich after checking, these PRs should fix the issue : https://github.com/ros-controls/ros2_control/pull/1175 and https://github.com/ros-controls/ros2_control/pull/1271

christophfroehlich, 2024-01-07:
Thanks, I'll check again if  the two PRs got merged!

closed 2024-01-12 by destogl