ros-controls/ros2_control #1110: how to find all controller？  ex.  forward_position_controller

zouzhe1, 2023-09-08:
I would like to ask how to find all the controllers supported by ros2 control and the format of controller information. 

For example, the forward_position_controller, 

I cannot find any documentation about it. 

In your official documentation at [https://control.ros.org/master/doc/ros2_controllers/doc/controllers_index.html#available-controllers](https://control.ros.org/master/doc/ros2_controllers/doc/controllers_index.html#available-controllers), 
this controller is not listed. 

Moreover, when I click on each controller within this link, 
there are only a few sentences of introduction, 
without specific information or usage instructions.

saikishor, 2023-09-08:
Hello @zouzhe1,

You can find the ros2_controllers in the repository: https://github.com/ros-controls/ros2_controllers, you can find the code to check it out. Moreover, some of those controllers use cases are showcased in the ros2_control_demos repository: https://github.com/ros-controls/ros2_control_demos. Check out these demos and the code, and you could contribute to the documentation by opening a PR.

Best Regards,

zouzhe1, 2023-09-09:
@saikishor 

Thank you for your answer. I have seen your links. I want to ask, in the YAML file, for example, the following code

```
123_controller:
      type: joint_trajectory_controller/JointTrajectoryController
456_controller:
      type: displacement_controller/DisplacementController
```

I know that 123_controller is a name chosen by the user, and type is the controller type provided by ROS control official, then I want to ask, where should I go to see this controller provided by the official? For example, there is no such controller as **displacement_controller** in all the links.

Where should I go to obtain this information? And for the usage of controllers.