ros-controls/ros2_control #1121: list_controllers is not returning

ravijo, 2023-09-24:
**Describe the bug**
The command `ros2 control list_controllers` is not returning anything and stays alive forever as if it hang!

**To Reproduce**
```
 ros2 control list_controllers
^C^C^C^C
^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C
```

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04.3 LTS
 - Version: Humble (Binary install)
- Using Cyclone DDS (As suggested by MoveIt)
- Have ROS Iron as well (But not source)

bmagyar, 2023-09-24:
Could you also give me versions of controller_manager and ros2controlcli? From the package.xml-s

ravijo, 2023-09-24:
@bmagyar 

I restarted the workstation and not it resturns properly. 

```
$ ros2 control list_controllers 
Could not contact service /controller_manager/list_controllers
```

This issue can be closed! Thank you very much.

closed 2023-09-24 by bmagyar

ravijo, 2023-09-26:
@bmagyar 

I came accross the same issue again. I noticed that somehow when I am invoking the `ur_robot_driver`, the ``ros2 control list_controllers ` does not return anything and stays alive forever.
```
$ ros2 launch ur_robot_driver ur5e.launch.py robot_ip:=aaa.bbb.ccc.ddd use_fake_hardware:=true initial_joint_controller:=joint_trajectory_controller launch_rviz:=false
...
...
...
[ros2_control_node-1] 2023-09-26 21:06:47.155 [RTPS_READER_HISTORY Error] Change payload size of '24' bytes is larger than the history payload size of '11' bytes and cannot be resized. -> Function can_change_be_added_nts
```

### Version Info
* ros-humble-controller-manager: 2.31.0-1jammy.20230919.235447
* ros-humble-ros2-control: 2.31.0-1jammy.20230920.001409

PS: It may be related to [this issue](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/823).