ros-controls/ros2_control #1122: Write commands not updated after some time

avanmalleghem, 2023-09-28:
I'm using `ros2_control` with `diff_drive_controller` and a custom `hardware_interface`. Here is the write method of this custom interface.

  ```cpp
hardware_interface::return_type MyRobot::write(
  const rclcpp::Time & /*time*/,
  const rclcpp::Duration & /*period*/)
{
  RCLCPP_DEBUG_STREAM_THROTTLE(
    rclcpp::get_logger(
      "otto_hw_interface"), steady_clock_, 5000,
    "command left : " << hw_commands_[0] << " (rad/s) - command right : " << hw_commands_[1] <<
      " (rad/s)");

  return hardware_interface::return_type::OK;
}
```

The issue is that after some time (randomly, sometimes more than 1hour) running correctly, I observe that `hw_commands_` is not populated anymore.

Can you help me understand this behavior ?

Thanks in advance.

**Environment (please complete the following information):**
 - ROS2 Humble
 - Yocto Distro
 - Jetson Nano
 - diff_drive_controller : 2.23.0
 - controller_manager : 2.28.0

bmagyar, 2023-09-28:
By not populated you mean you get the same prints or that the prints stop?

Are you sure the controller's update function is called? You could check
that the controller is still active and receiving commands

avanmalleghem, 2023-09-28:
Hello, 

I receive 0 during like 10 minutes and then I receive correct values again.
I've observed that it is because I enter into the time out condition of the update method in the diff_drive_controller.
I'm currently troubleshooting it and come back to you at each step.

But of course, if you have any idea based on your experience, I'm taking it !!!

bmagyar, 2023-09-28:
I suspect this is controller behaviour. If the diff drive controller times
out, it will stop the robot, hence the zeroes.

avanmalleghem, 2023-09-29:
I went a little further... When I `ros2 topic echo` from my host on the cmd_vel topic (just in front of the diff_drive_controller), I see messages in it but I still see timeout in the diff_drive_controller (see my screen where I added logs in the diff_drive_controller code, you see the growing value of the variable "age_of_last_command"). If I `ros2 topic echo` from the jetson nano on the same cmd_vel topic, the diff_drive_controller starts working again and "age_of_last_command" variable is more or less 0 again...

[image]

avanmalleghem, 2023-10-05:
Update : it looks like it is linked to FastDDS. When I switch to CycloneDDS, it works like a charm.

bmagyar, 2023-10-05:
Oh dear... 🤦 So is it something where the dif drive doesn't get more messages?

avanmalleghem, 2023-10-10:
@bmagyar nop, it happens whatever you do but after some times (it can be hours). The behavior is that, after some times, the diff_drive_controller doesn't consume input cmd_vel anymore when using FastDDS.

closed 2025-05-19 by github-actions[bot]