ros-controls/ros2_control #1128: hardware_components_initial_state doesn't work

MattiaDeiRossi, 2023-10-06:
Hi all, 

I would to launch `ros2_control_node` having my hardware component on `inactive` state as first state.
I read the documentation at [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html](https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html) that works like

```
hardware_components_initial_state:
  unconfigured:
    - "arm1"
    - "arm2"
  inactive:
    - "base3"
```

and my ros2_control description looks like:

```
<?xml version="1.0" encoding="utf-8"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">
  <xacro:macro name="awtube_ros2_control"
    params="use_fake_hardware:=^|true robot_id:=^|awtube brand:=^|synapticon mode_of_op:=^|8">

    <ros2_control name="AwTubeSystem" type="system">
      <hardware>
        <xacro:if value="${use_fake_hardware}">
          <plugin>fake_components/GenericSystem</plugin>
        </xacro:if>
        <xacro:unless value="${use_fake_hardware}">
          <plugin>ethercat_driver/EthercatDriver</plugin>
          <param name="master_id">0</param>
          <param name="control_frequency">1000</param>
        </xacro:unless>
      </hardware>
      .
      .
      .
    </ros2_control>
  </xacro:macro>
</robot>
```

When writing in my `ros2_controller.yam` the following code, controllers go immediately in the activate state, and do not stop in the inactive. What am I doing wrong?
```
controller_manager:
  ros__parameters:
    update_rate: 1000
    hardware_components_initial_state:
      inactive:
      - AwTubeSystem
```

Thanks

christophfroehlich, 2023-10-09:
Hi! Which ROS distro are you using? we renamed fake_components already some time ago and use mock components from humble on.
If you are on <=galactic: We haven't backported the parameter you are trying to use. Be aware of the different versions of the documentation for different releases, there is also a difference of this feature/parameters between humble and rolling.

MattiaDeiRossi, 2023-10-10:
Hi, 
Thank you for your answer.
I' m using `humble` distro. I tried to change `<plugin>fake_components/GenericSystem</plugin>` into `<plugin>mock_components/GenericSystem</plugin>` but the behaviour is the same as before.

Could you provide a very simple application example please?

christophfroehlich, 2023-10-10:
Have a look on [this example](https://control.ros.org/master/doc/ros2_control_demos/example_2/doc/userdoc.html) how to use the mock_component, but this is not the problem here. (you should have got a warning with humble, that the fake_component is deprecated?).

You write that your controller goes into active state? But the parameter is handling the hardware components.. I'm not sure what the controller_manager is doing with the controller, if the hw component is not in active state.

Try `ros2 control list_hardware_components`, what does it tell you?

MattiaDeiRossi, 2023-10-10:
I have a look on the  [example2](https://control.ros.org/master/doc/ros2_control_demos/example_2/doc/userdoc.html) using `humble` version on docker

I haven't any issue if I change 
```
     <hardware>
        <plugin>ros2_control_demo_example_2/DiffBotSystemHardware</plugin>
        <param name="example_param_hw_start_duration_sec">0</param>
        <param name="example_param_hw_stop_duration_sec">3.0</param>
      </hardware> 
```
with
```
      <hardware>
        <plugin>fake_components/GenericSystem</plugin>
        <param name="calculate_dynamics">true</param>
      </hardware>
```
and the same behaviour is using mock_components/GenericSystem.

So regarding the initial state I changed `diffbot_controllers.yaml` as following
```
controller_manager:
  ros__parameters:
    update_rate: 10  # Hz
    hardware_components_initial_state:
      unconfigured:
        - "DiffBot"

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    diffbot_base_controller:
      type: diff_drive_controller/DiffDriveController
...
```
and `diffbot.ros2_control.xacro` with
```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

  <xacro:macro name="diffbot_ros2_control" params="name prefix">

    <ros2_control name="${name}" type="system">
      <!-- <hardware>
        <plugin>ros2_control_demo_example_2/DiffBotSystemHardware</plugin>
        <param name="example_param_hw_start_duration_sec">0</param>
        <param name="example_param_hw_stop_duration_sec">3.0</param>
      </hardware> -->
      <hardware>
        <plugin>mock_components/GenericSystem</plugin>
        <param name="calculate_dynamics">true</param>
      </hardware>
      <joint name="${prefix}left_wheel_joint">
        <command_interface name="velocity"/>
        <state_interface name="position"/>
        <state_interface name="velocity"/>
      </joint>
      <joint name="${prefix}right_wheel_joint">
        <command_interface name="velocity"/>
        <state_interface name="position"/>
        <state_interface name="velocity"/>
      </joint>
    </ros2_control>
  </xacro:macro>
</robot>
```
I expected Diffbots hardware to be in an unconfigured state, but the command shows
```
root@docker-desktop:/home/ros2_ws# ros2 launch ros2_control_demo_example_2 diffbot.launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-10-10-20-30-09-135514-docker-desktop-17079
[... 27 lines cut ...]
[spawner-3] [INFO] [1696969809.956563556] [spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster
[INFO] [spawner-3]: process has finished cleanly [pid 17095]
[INFO] [spawner-4]: process started with pid [17150]
[INFO] [rviz2-5]: process started with pid [17152]
[rviz2-5] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[spawner-4] [INFO] [1696969812.295427692] [spawner_diffbot_base_controller]: Waiting for '/controller_manager' node to exist
[ros2_control_node-1] [INFO] [1696969813.305118157] [controller_manager]: Loading controller 'diffbot_base_controller'
[spawner-4] [INFO] [1696969813.356438410] [spawner_diffbot_base_controller]: Loaded diffbot_base_controller
[ros2_control_node-1] [INFO] [1696969813.357631381] [controller_manager]: Configuring controller 'diffbot_base_controller'
[spawner-4] [INFO] [1696969813.556429637] [spawner_diffbot_base_controller]: Configured and activated diffbot_base_controller
[INFO] [spawner-4]: process has finished cleanly [pid 17150]`
```
and `ros2 control list_hardware_components` 
```
root@docker-desktop:/home/ros2_ws# ros2 control list_hardware_components
Hardware Component 1
        name: DiffBot
        type: system
        plugin name: mock_components/GenericSystem
        state: id=3 label=active
        command interfaces
                left_wheel_joint/velocity [available] [claimed]
                right_wheel_joint/velocity [available] [claimed]
```
that seems that hardware component is on active state

christophfroehlich, 2023-10-11:
The problem is that the parameter got renamed in rolling but the [backport got stuck](https://github.com/ros-controls/ros2_control/pull/1064). Have a look at the docs of the [humble branch](https://control.ros.org/humble/doc/ros2_control/controller_manager/doc/userdoc.html#parameters), you have to use `configure_components_on_start` unless the changes will be backported.

closed 2023-10-12 by MattiaDeiRossi

christophfroehlich, 2023-10-12:
you're welcome, does it work now?

MattiaDeiRossi, 2023-10-12:
It works now as expected and the `hardware` component is in the desired state.
In this way happens that `ros2_control_node` call read() and write() functions automatically and my hardware plugin crashes because it needs to have allocated all interfaces in read() and write().

1) Is it correct behaviour that read() and write() methods are invoked if hardware is in a different state that `active` ?
2) Once the hardware is started in a different state than `active`, How is it possible to switch its state?

christophfroehlich, 2023-10-12:
1. yes this is intentionally, but confused me in the beginning, too. 
2. There is an old [PR](https://github.com/ros-controls/ros2_control_demos/pull/187) for the demos concerning the lifecycle management, updated with this [PR](https://github.com/StoglRobotics-forks/ros2_control_demos/pull/7). Have a look at [this section here](https://github.com/StoglRobotics-forks/ros2_control_demos/blob/9fac0836375a2a904c5645b9f0acc6b1241ae0ca/README.md#example-xx-multi-robot-system-with-tools), you have to manually call a service of the controller_manager because there is no CLI verb yet.

saikishor, 2024-10-08:
> I finally found this issue. But I am still with a similar problem. I need the hardware to be unconfigured on start. I set the hardware_components_initial_state but every time the resource manager configure and activate the hardware automatically. I also checked the source code (Humble) at https://github.com/ros-controls/ros2_control/blob/humble/controller_manager/src/controller_manager.cpp but find out nothing. It seems the code does not work.

@Elfits you can set the components to the unconfigured state using this parameter `hardware_components_initial_state.unconfigured`

https://github.com/ros-controls/ros2_control/blob/b00988134ddfe2e3fbd8178124fd41981b84ae41/controller_manager/src/controller_manager.cpp#L406-L409