ros-controls/ros2_control #1133: ros2controlcli: use-sim-time

christophfroehlich, 2023-10-10:
With #1132 I also saw now that the help shows two options coming from ros2cli

```bash
  --spin-time SPIN_TIME
                        Spin time in seconds to wait for discovery (only applies when not using an already running daemon)
  -s, --use-sim-time    Enable ROS simulation time
```

The latter is sometimes part of the docs, sometimes not. [it was introduced 2 years ago](https://github.com/ros2/ros2cli/pull/581), that's maybe why.

As far as I understand ros2cli, it creates a node which then is used for calling the services with our ros2controlcli? Does it make a difference if use-sim-time is set for the service client, or not?

I'd correct/complete the docs but want to understand it first.

saikishor, 2025-05-19:
I don't think it makes a huge difference when it deals with the services

christophfroehlich, 2025-05-19:
Then we can close this issue. We can't remove the options anyways, and the docs were updated in the meantime

closed 2025-05-19 by christophfroehlich