ros-controls/ros2_control #1136: Cannot unload unconfigured controller

firesurfer, 2023-10-13:
**Describe the bug**

When having an erronous controller which can not be configured it is left in the state `unconfigured`.
When trying to unload that controller there is no rcl life cycle transition found

``
[ros2_control_node-11] [WARN] [1697180791.709508944] [rcl_lifecycle]: No transition matching 2 found for current state unconfigured
[ros2_control_node-11] [ERROR] [1697180791.709542087] [cartesian_compliance_controller_top]: Unable to start transition 2 from current state unconfigured: Transition is not registered., at ./src/rcl_lifecycle.c:355
```

**To Reproduce**

1. Have a controller which cannot be configured
2. Run `ros2 control  unload_controller my_bad_controller`

**Expected behavior**

Being able to unload the controller

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Version: Iron binary install

firesurfer, 2024-02-01:
@saikishor  or someone else. Sorry to push. But I think this issue should not be forgotten.

saikishor, 2024-02-01:
@firesurfer thanks for the reminder. If I have some time during this or next week, I'll try to fix it

saikishor, 2024-02-01:
@firesurfer I tried to reproduce your error. I see the same error message but at the end of the day, it does get removed from the controller list. It's just that the node cleanup is only needed if the controller is in an inactive state, if it is in unconfigured state, this won't be needed.

saikishor, 2024-02-01:
@firesurfer  I've created a pull request with the fix

firesurfer, 2024-02-02:
@saikishor  Okey good to know. I just tried to unload the controller and saw the error message. 
Great that you already fixed it!

saikishor, 2024-02-02:
@firesurfer can test the PR https://github.com/ros-controls/ros2_control/pull/1363 and if it works, please approve it. So, we can get it merged and backport it

closed 2024-02-09 by bmagyar