ros-controls/ros2_control #1143: What is the use of controller_manager ros2_control_node

SmnGdn, 2023-10-23:
Hi, 

I have a project in which I have different controllers (tricycle and joint state broadcaster) from the package ros2_controllers to simulate my robot in gazebo. 
I am just loading the yaml files of these controllers in my URDF, inside a gazebo_ros2_control plugin. Then  I'm spawning these controllers using a controller_manager  spawner node. I see that in all the ros2_control_demos examples, the node controller_manager ros2_control_node is also added, but in my case everything works without it. 

I guess I'm missing something but not sure what. Is it that this node is only needed when not using a simulator?

Thanks in advance!

bmagyar, 2023-10-23:
You got it right!

The ros2_control_node implements the control loop which could be summed up
as a loop of read-uodate-write plus some extra work.

In the case of gazebo, the gazebo plugin implements the control loop since
it is the entity controlling time. We should probably have a whole section
on this in the docs somewhere... Or we may already have!

SmnGdn, 2023-10-23:
Thank you for your answer! Very clear :)

I don't know if there is a section about it in the docs, but I can definitely say I looked for it and didn't find any.

closed 2023-11-03 by SmnGdn