ros-controls/ros2_control #1144: Mock components cannot execute MoveIt trajectories for robots with multiple command interfaces

sea-bass, 2023-10-24:
After merging https://github.com/ros-controls/ros2_control/pull/1028, our default `moveit2_tutorials` demo is no longer able to execute trajectories with the following error.

```
[ros2_control_node-5] [ERROR] [1698167176.237792390] [mock_generic_system]: Got multiple (2) starting interfaces for joint 'joint_1' - this is not supported!
[ros2_control_node-5] [ERROR] [1698167176.237800951] [mock_generic_system]: Got multiple (2) starting interfaces for joint 'joint_2' - this is not supported!
[ros2_control_node-5] [ERROR] [1698167176.237804701] [mock_generic_system]: Got multiple (2) starting interfaces for joint 'joint_3' - this is not supported!
[ros2_control_node-5] [ERROR] [1698167176.237807725] [mock_generic_system]: Got multiple (2) starting interfaces for joint 'joint_4' - this is not supported!
[ros2_control_node-5] [ERROR] [1698167176.237810902] [mock_generic_system]: Got multiple (2) starting interfaces for joint 'joint_5' - this is not supported!
[ros2_control_node-5] [ERROR] [1698167176.237814015] [mock_generic_system]: Got multiple (2) starting interfaces for joint 'joint_6' - this is not supported!
[ros2_control_node-5] [ERROR] [1698167176.237817238] [mock_generic_system]: Got multiple (2) starting interfaces for joint 'joint_7' - this is not supported!
[ros2_control_node-5] [ERROR] [1698167176.237828626] [resource_manager]: Component 'KortexMultiInterfaceHardware' did not accept new command resource combination: 
[ros2_control_node-5]  Start interfaces: 
[ros2_control_node-5] [
[ros2_control_node-5]   joint_1/position
[ros2_control_node-5]   joint_1/velocity
[ros2_control_node-5]   joint_2/position
[ros2_control_node-5]   joint_2/velocity
[ros2_control_node-5]   joint_3/position
[ros2_control_node-5]   joint_3/velocity
[ros2_control_node-5]   joint_4/position
[ros2_control_node-5]   joint_4/velocity
[ros2_control_node-5]   joint_5/position
[ros2_control_node-5]   joint_5/velocity
[ros2_control_node-5]   joint_6/position
[ros2_control_node-5]   joint_6/velocity
[ros2_control_node-5]   joint_7/position
[ros2_control_node-5]   joint_7/velocity
[ros2_control_node-5] ]
[ros2_control_node-5] Stop interfaces: 
[ros2_control_node-5] [
[ros2_control_node-5] ]
[ros2_control_node-5] 
[ros2_control_node-5] [ERROR] [1698167176.237834419] [controller_manager]: Could not switch controllers since prepare command mode switch was rejected.
[spawner-7] [ERROR] [1698167176.238412335] [spawner_joint_trajectory_controller]: Failed to activate controller
```

We can get around it by ensuring that our `ros2_controllers.yaml` file does not specify velocity command interfaces, which works for our Kinova Gen3 setup since it doesn't support velocity anyway... but this might break similar setups for users that did previously have JTCs or other controllers with position + velocity interfaces, for example.

bmagyar, 2023-10-24:
Funny thing is that I think I even encountered this with the workshop
content

destogl, 2023-11-01:
OK, I know the problem here and know how to solve it. Unfortunately, it is not an easy fix, since we have to define what this actually means when one sets “velocity” and “position” at the same time. I am planning to do the following in that case in the mock hardware:

**When using “position” and “velocity” interfaces on the same joint – the joint will move to the position with the given velocity and stop there, even if the velocity is still set to a value different from 0.**

If there are other simple ideas like this, I am open to listening to them :)

sea-bass, 2023-11-01:
I think this is fine -- we could issue a (throttled) warning that the velocity is basically being ignored.

PR #1151 (2023-11-06): [MockHardware] Fix the issues where hardware with multiple interfaces can not be started because of a logical bug added when adding dynamics calculation functionality.
Fixes #1144.

I have introduces this bug. The issues is that if one is not using dynamic calculations there is no limitation on interfaces used.

I have also added a test to avoid this hapening again.
closed 2023-11-06 by destogl