ros-controls/ros2_control #1162: Colcon build issues

sandropp, 2023-11-12:
**Describe the bug**
A clear and concise description of what the bug is.
when execute 'colcon build', the following error:
Finished <<< transmission_interface [48.4s]                                             
--- stderr: controller_manager                                    
/home/sandro/ros2_control_ws/src/ros2_control/controller_manager/src/controller_manager.cpp: In member function ‘void controller_manager::ControllerManager::list_hardware_components_srv_cb(std::shared_ptr<controller_manager_msgs::srv::ListHardwareComponents_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ListHardwareComponents_Response_<std::allocator<void> > >)’:
/home/sandro/ros2_control_ws/src/ros2_control/controller_manager/src/controller_manager.cpp:1777:43: error: ‘std::tuple_element<1, const std::pair<const std::__cxx11::basic_string<char>, hardware_interface::HardwareComponentInfo> >::type’ {aka ‘const struct hardware_interface::HardwareComponentInfo’} has no member named ‘class_type’
 1777 |     component.class_type = component_info.class_type;
      |                                           ^~~~~~~~~~
/home/sandro/ros2_control_ws/src/ros2_control/controller_manager/src/controller_manager.cpp: In lambda function:
/home/sandro/ros2_control_ws/src/ros2_control/controller_manager/src/controller_manager.cpp:2336:7: warning: control reaches end of non-void function [-Wreturn-type]
 2336 |       };
      |       ^
gmake[2]: *** [CMakeFiles/controller_manager.dir/build.make:76: CMakeFiles/controller_manager.dir/src/controller_manager.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:208: CMakeFiles/controller_manager.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< controller_manager [13.9s, exited with code 2]

[image]

**Environment (please complete the following information):**
 - OS: ubuntu 22.04
 - Version : humble
- Anything that may be unusual about your environment

**Additional context**
Add any other context about the problem here, especially include any modifications to ros2_control that relate to this issue.

bmagyar, 2023-11-12:
> std::__cxx11::basic_string, hardware_interface::HardwareComponentInfo>

Since you haven't given us much to work with, I'm going to guess you are
trying to build the master branch on humble.

This won't work. Please use the humble branch for humble. Even better,
don't build ros2_control from source but just install it from binaries with
apt.

sandropp, 2023-11-12:
thanks, after switching to humble branch, it is ok now.

closed 2023-11-12 by bmagyar