ros-controls/ros2_control #1170: Error when loading multiple controllers (controller_sorting)

GilmarCorreia, 2023-11-16:
**Bug:**
I am utilizing ros2_control to integrate a custom simulator with the ros2_control framework, and I have encountered an issue when using a robot that has 20 `range_sensor_broadcasters`, 1 `joint_state_broadcaster`, and 1 `velocity_controller`. During the attempt to configure controllers, everything proceeds without a problem up to 16 controllers (e.g., 1 `joint_state_broadcaster`, 1 `velocity_controller`, and 14 `range_sensor_broadcasters`). However, when I modify the robot's .yaml file to configure 17 controllers (e.g., 1 `joint_state_broadcaster`, 1 `velocity_controller`, and 15 `range_sensor_broadcasters`), the following error occurs:

```bash
...
[INFO] [1700161750.904241702] [PioneerV2_237.controller_manager]: Loading controller 'Pioneer_p3dx_ultrasonicSensor15'
[INFO] [1700161750.912386985] [PioneerV2_237.controller_manager]: Configuring controller 'Pioneer_p3dx_ultrasonicSensor15'

/home/MY_USER/Desktop/ros2_ws/install/controller_manager/lib/libcontroller_manager.so(_ZN18controller_manager17ControllerManager18controller_sortingERKNS_14ControllerSpecES3_RKSt6vectorIS1_SaIS1_EE+0x5c)[0x7f9ba1fdb550]
/home/MY_USER/Desktop/ros2_ws/install/controller_manager/lib/libcontroller_manager.so(_ZSt13__invoke_implIbRMN18controller_manager17ControllerManagerEFbRKNS0_14ControllerSpecES4_RKSt6vectorIS2_SaIS2_EEERPS1_JRS2_SF_RS7_EET_St21__invoke_memfun_derefOT0_OT1_DpOT2_+0xb1)[0x7f9ba20714a4]
/home/MY_USER/Desktop/ros2_ws/install/controller_manager/lib/libcontroller_manager.so(_ZSt8__invokeIRMN18controller_manager17ControllerManagerEFbRKNS0_14ControllerSpecES4_RKSt6vectorIS2_SaIS2_EEEJRPS1_RS2_SF_RS7_EENSt15__invoke_resultIT_JDpT0_EE4typeEOSI_DpOSJ_+0x83)[0x7f9ba205d61d]
/home/MY_USER/Desktop/ros2_ws/install/controller_manager/lib/libcontroller_manager.so(_ZNSt5_BindIFMN18controller_manager17ControllerManagerEFbRKNS0_14ControllerSpecES4_RKSt6vectorIS2_SaIS2_EEEPS1_St12_PlaceholderILi1EESD_ILi2EES7_EE6__callIbJRS2_SJ_EJLm0ELm1ELm2ELm3EEEET_OSt5tupleIJDpT0_EESt12_Index_tupleIJXspT1_EEE+0xe1)[0x7f9ba204d75d]
corrupted double-linked list
```

**Environment:**
- OS: Windows 11 (WSL Ubuntu-22.04)
- Version: Humble
- Language: C++

**Additional Context:**
I have noticed that the error occurs within the code in `controller_manager.cpp` at line 818:

```cpp
// Reordering the controllers
std::sort(
  to.begin(), to.end(),
  std::bind(
    &ControllerManager::controller_sorting, this, std::placeholders::_1, std::placeholders::_2,
    to));
```

Removing these lines has resolved the issue for me. However, I am unsure how to fix the `controller_sorting` function.

saikishor, 2023-11-17:
Hello @GilmarCorreia!

Thank you for reporting the issue. I haven't tested with 17 controllers till now for sure, I am amazed that it fails exactly with 17 controllers. The maximum number of controllers I tested with is around 10-11 controllers.  I have to take a look at the issue. I will try to write some tests to reproduce and fix it in the upcoming days. Sorry for the inconvenience caused.

Thank you,

Best Regards,
Sai

saikishor, 2023-11-24:
Hello @GilmarCorreia!

I have a fix for the crash that you are facing. You can find it in the above linked PR. It works for the complex test cases that I've added. It would be great if you could test the changes on your setup and confirm that this solution is stable enough for you.

Thank you,

Best Regards,
Sai

closed 2023-12-03 by saikishor

GilmarCorreia, 2023-12-12:
Sorry for the delay in response, I have the updates downloaded now, I just need some time to test. Thank you very much for your attention and time solving the issue @saikishor!