ros-controls/ros2_control #1177: Issue using more than one hardware interface to control 2 robots

Zarnack, 2023-11-23:
**Describe the bug**
I want to use two robot arms (Franka Emika Panda) at the same time with Moveit 2 and ros2_control. For this I have set up two different hardware components called:
- panda_3_FrankaHardwareInterface
  - with command/state interface like this:  _panda_3_joint1/...._ 
  - with the _panda_3_arm_controller_ of type _joint_trajectory_controller/JointTrajectoryController_
- panda_1_FrankaHardwareInterface
  - with command/state interface like this:  _panda_1_joint1/...._ 
  - with the _panda_1_arm_controller_ of type _joint_trajectory_controller/JointTrajectoryController_

Using only the _fake_components/GenericSystem_ everything works just fine. I can control both robots and no errors occur.
Using the _franka_hardware/FrankaHardwareInterface_ the problems start:
- Everything works with only one robot (e.g. only using _panda_1_arm_controller_)
- using a second robot the initialization of the second controller (_panda_3_arm_controller_) fails:

```
[ros2_control_node-4] [INFO] [1700586290.750446657] [FrankaHardwareInterface]: Preparing command mode switch
[ros2_control_node-4] [INFO] [1700586290.750465133] [FrankaHardwareInterface]: ++ panda_3_joint1/effort
[ros2_control_node-4] [INFO] [1700586290.750468434] [FrankaHardwareInterface]: ++ panda_3_joint2/effort
[ros2_control_node-4] [INFO] [1700586290.750470715] [FrankaHardwareInterface]: ++ panda_3_joint3/effort
[ros2_control_node-4] [INFO] [1700586290.750472862] [FrankaHardwareInterface]: ++ panda_3_joint4/effort
[ros2_control_node-4] [INFO] [1700586290.750475009] [FrankaHardwareInterface]: ++ panda_3_joint5/effort
[ros2_control_node-4] [INFO] [1700586290.750477177] [FrankaHardwareInterface]: ++ panda_3_joint6/effort
[ros2_control_node-4] [INFO] [1700586290.750479193] [FrankaHardwareInterface]: ++ panda_3_joint7/effort
[ros2_control_node-4] [ERROR] [1700586290.750490794] [FrankaHardwareInterface]: Switching between control modes without stopping it first is not supported.
[ros2_control_node-4] Please stop the running controller first.
[ros2_control_node-4] [ERROR] [1700586290.750508309] [resource_manager]: Component 'panda_1_FrankaHardwareInterface' did not accept new command resource combination: 
[ros2_control_node-4]  Start interfaces: 
[ros2_control_node-4] [
[ros2_control_node-4]   panda_3_joint1/effort
[ros2_control_node-4]   panda_3_joint2/effort
[ros2_control_node-4]   panda_3_joint3/effort
[ros2_control_node-4]   panda_3_joint4/effort
[ros2_control_node-4]   panda_3_joint5/effort
[ros2_control_node-4]   panda_3_joint6/effort
[ros2_control_node-4]   panda_3_joint7/effort
[ros2_control_node-4] ]
[ros2_control_node-4] Stop interfaces: 
[ros2_control_node-4] [
[ros2_control_node-4] ]
[ros2_control_node-4] 
[ros2_control_node-4] [ERROR] [1700586290.750512947] [controller_manager]: Could not switch controllers since prepare command mode switch was rejected.
```
The _join_state_broadcaster_ still works fine for both robots and I can control the first robot without any problems.
It seems like it is trying to start the panda_3_... interfaces on the wrong hardware component (_panda_1_FrankaHardwareInterface_ instead of _panda_3_FrankaHardwareInterface_) when activating the panda_3_arm_controller.

**Expected behavior**
The second controller should start succesfully. The controller manager should try to start the panda_3_... interfaces on the panda_3_FrankaHardwareInterface' and not on the panda_1_FrankaHardwareInterface'.

**Questions**
Am I doing something wrong here? Do I need to make changes to the hardware interface (e.g. some namespacing stuff) to have it working with multiple instances of itself or is this just a bug?

**Environment:**
 - ROS2 Humble in Docker with Ubuntu 22.04
 - tested with ros2_control installed from binaries and from source
 - franka_hardware/FrankaHardwareInterface from [here](https://github.com/tenfoldpaper/panda_ros2/tree/humble/franka_hardware)

**detailed Setup**
panda_ros_controllers.yaml:
```
controller_manager:
  ros__parameters:
    update_rate: 1000  # Hz

    panda_1_arm_controller:
      type: joint_trajectory_controller/JointTrajectoryController
    panda_2_arm_controller:
      type: joint_trajectory_controller/JointTrajectoryController
    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

[... 136 lines cut ...]
        state: id=3 label=active
        command interfaces
                panda_3_joint1/effort [available] [unclaimed]
                panda_3_joint1/position [available] [unclaimed]
                panda_3_joint1/velocity [available] [unclaimed]
                panda_3_joint2/effort [available] [unclaimed]
                panda_3_joint2/position [available] [unclaimed]
                panda_3_joint2/velocity [available] [unclaimed]
   ....
Hardware Component 2
        name: panda_1_FrankaHardwareInterface
        type: system
        plugin name: franka_hardware/FrankaHardwareInterface
        state: id=3 label=active
        command interfaces
                panda_1_joint1/effort [available] [claimed]
                panda_1_joint1/position [available] [unclaimed]
                panda_1_joint1/velocity [available] [unclaimed]
                panda_1_joint2/effort [available] [claimed]
                panda_1_joint2/position [available] [unclaimed]
                panda_1_joint2/velocity [available] [unclaimed]
....
```

Any help or tips will be appreciated!

Zarnack, 2024-02-20:
This seems to be a problem with moveit2 and using the default resource manager logic of ros2_control. When I switch the controller manager to the moveit_simple_controller_manager/MoveItSimpleControllerManager this problem disappears.

To change, add this to your yaml describing the controller structure for moveit:

```
moveit_manage_controllers: True
moveit_controller_manager: moveit_simple_controller_manager/MoveItSimpleControllerManager
```

closed 2024-02-20 by Zarnack

karanchahal, 2024-07-13:
I think I am facing this same problem, I have 2 controllers loaded but 1 of which is complaing that the joint is not found.

I have 2 different ros2_control tags with different topic based controls for each.

pratyush-p, 2024-07-21:
> This seems to be a problem with moveit2 and using the default resource manager logic of ros2_control. When I switch the controller manager to the [...]

Can you specify where exactly in the yaml file it should be kept?