ros-controls/ros2_control #1179: Switch controllers services return always ok

halejo-as, 2023-11-24:
**Describe the bug**
Calling the service **/controller_manager/switch_controller** return always true even if it could not switch the controllers.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch any demo with a set of controllers loaded
2. Switch for a non existing controller 

```
ros2 service call /controller_manager/switch_controller controller_manager_msgs/srv/SwitchController "{start_controllers: ["dummy_controller"]}"
```

**Expected behavior**
The response expected should be ok = False but the ok is equals to true  

```
response:
controller_manager_msgs.srv.SwitchController_Response(ok=True)
```

The controller manager node notify there is an error:

Resource conflict for controller 'dummy_controller'. Command interface 'XXXX' is already claimed.

**Environment (please complete the following information):**
 - Ubuntu 20.04
 - Foxy

Thanks for your help

bmagyar, 2023-11-24:
Foxy is End-Of-Life officially, there will be no more builds or releases for it.
Could you please try reproducing the same on Humble/Iron/Rolling?

halejo-as, 2024-01-07:
Hello, 

I  tested in humble and appears the same problem:
1.  Launch the controllers, for example, ` ros2_control_demo_example_1 rrbot.launch.py`
2. Switch for a non existing controller 

**RESULT:** 

The controller manager notify the error
```
[controller_manager]: Could not 'activate' controller with name 'dummy_controller' because no controller with this name exists      
[controller_manager]: Empty activate and deactivate list, not requesting switch             
``` 
But the service response is Ok
 ```          
response:
controller_manager_msgs.srv.SwitchController_Response(ok=True)
```

gagapaga, 2024-01-26:
same problem

saikishor, 2024-01-26:
Hello!

Please try with the `strictness` set to `STRICT` when sending the service request. I think it should respond with false.

Thank you

christophfroehlich, 2024-01-26:
Hint: the problem with `STRICT` is, that it does return false also if the controller exists but is already in the desired state.

closed 2025-05-19 by github-actions[bot]