ros-controls/ros2_control #1183: [Diagnostics] - A non active controller should not be reported as error

firesurfer, 2023-11-30:
**Describe the bug**

Currently controllers that are loaded but are deactivated are reported as errors via ROS diagnostics. 
[image]

I think this should be a warning. In my system for example I regularly switch controllers (e.g. from a compliant controller to the jtc and back). The other controller being inactive is desired behavior in that case and should not be reported as an error.

**To Reproduce**

Load a controller and deactivate it

**Expected behavior**

Report nothing or just a warning via ROS diagnostics

**Environment (please complete the following information):**
 - OS:  Ubuntu 22.04
 - Version iron

firesurfer, 2023-11-30:
@saikishor I created a PR which just makes the switch from Error to Warn. Not sure if there is any other error information to report  about controller (such as ros2 control could not active the controller)

christophfroehlich, 2023-12-05:
Is this fixed with #1184?

saikishor, 2023-12-05:
@christophfroehlich yes it should be fixed. Now it won't report it as an error

closed 2023-12-05 by christophfroehlich