ros-controls/ros2_control #1185: /controller_manager/list_controllers service is taking too long to respond

samialperen, 2023-12-02:
**Describe the bug**
/controller_manager/list_controllers service is taking around 600 ms to respond.
This is how I measure the rough time
`time ros2 service call /controller_manager/list_controllers controller_manager_msgs/srv/ListControllers`

I am not sure whether this is expected or not. If I am missing any kind of details during the setup, please let me know.  Thank you!  

**To Reproduce**
Steps to reproduce the behavior:
1. In a fresh workspace, launch [example_1 of ros2_control (https://control.ros.org/master/doc/ros2_control_demos/example_1/doc/userdoc.html)
2. Once the robot is up and running check the time for list controllers to take `time ros2 service call /controller_manager/list_controllers controller_manager_msgs/srv/ListControllers`
3. For me, it takes around 600 ms each time I run the above command

**Expected behavior**
I would expect this service to be faster. If that is not case, can someone explain why it is normal for this service to take this long? 

**Screenshots**
N/A

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04 - Docker container
 - Version: Humble

**Additional context**
N/A

bmagyar, 2023-12-02:
For completeness could you please also try 
`ros2 control list_controllers`?

Have you tried using *the other DDS*? (I never know what the default is)

samialperen, 2023-12-02:
@bmagyar Thanks a lot for responding quick Bence. 

```
time ros2 control list_controllers
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active    
forward_position_controller[forward_command_controller/ForwardCommandController] active    

real	0m1.094s
user	0m0.326s
sys	0m0.072s
```

I use cyclone_dds.

samialperen, 2023-12-05:
Any news on this @bmagyar ? We can also consider my question in a broader context. If you happen to see this kind of regression/slowness in ros2 control's services, what are the best practices or common troubleshooting guidelines?

bmagyar, 2023-12-05:
Have you tried FastDDS too?

bmagyar, 2023-12-05:
Hey, if it is any consolation, I get this on Rolling:

```
$ time ros2 control list_controllers
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active    
forward_position_controller[forward_command_controller/ForwardCommandController] active    

real	0m1.289s
user	0m0.912s
sys	0m1.267s
```

samialperen, 2023-12-06:
Fast DDS is also performing the same way. It is interesting that you also get a similar numbers for Rolling almost a second to get a response from that service. After checking [the service callback](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/src/controller_manager.cpp#L1582), I am not sure why it is taking that long considering in your example there are only 2 controllers.  
Is there any potential for us to improve the performance of this service? Do you see any easy to target code block in the service callback which might be causing downside in performance? If not, I will try to run a profiler for that code section to see which part is taking so long.

saikishor, 2023-12-06:
Hello!
I've just checked it, and the service callback itself is not taking time. If you see the timestamp in the log below from [the service callback](https://github.com/ros-controls/ros2_control/blob/b8ba5c0d0e13c322ad49d99c3d29367ed87714fe/controller_manager/src/controller_manager.cpp#L1582-L1675), it is processing it fast enough

```
[ros2_control_node-1] [DEBUG] [1701883834.987739589] [controller_manager]: list controller service called
[ros2_control_node-1] [DEBUG] [1701883834.987763516] [controller_manager]: list controller service locked
[ros2_control_node-1] [DEBUG] [1701883834.987781107] [controller_manager]: list controller service finished
```
However, It also takes around 1s for me as well through the command line

```
$ time ros2 control list_controllers
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active    
forward_position_controller[forward_command_controller/ForwardCommandController] active    

real	0m0.949s
user	0m0.304s
sys	0m0.049s
``` 
Atleast from the timestamps from the logs above, the problem is not inside the service callback

samialperen, 2023-12-08:
Yes, I can do that for sure, but we have multiple controllers, and we switch them on the fly and check whether they are active or not (they become deactivated in certain situations) and we need to check whether they are active or not (in certain situations) before moving the robot, so waiting 1 sec to do that is not desirable. However, if there is no workaround on that, that is totally fine. Just exploring options :)

bmagyar, 2023-12-09:
We are reporting the same info afaik on a diagnostics topic too

samialperen, 2024-01-10:
Thanks folks. :+1:

closed 2024-01-10 by samialperen

saikishor, 2025-10-18:
> > Hello [@wsxq2](https://github.com/wsxq2)
> > 
> > Are you compiling ros2_control locally? Or are you using binaries?
> 
> I use binaries.

I've tried to reproduce it these days, but couldn't. Could you time the load_controllers function? And if you could pinpoint the section of code, maybe we can take a look on how to improve.