ros-controls/ros2_control #1196: Qustion: ReadOnlyHandle

mhubii, 2023-12-07:
https://github.com/ros-controls/ros2_control/blob/b8ba5c0d0e13c322ad49d99c3d29367ed87714fe/hardware_interface/include/hardware_interface/handle.hpp#L32

Isn't the `ReadOnlyHandle` able to modify the pointed value? Shouldn't this be `const double* const`

yashi, 2024-01-13:
- `const double * value_ptr` means pointer to const double.  IOW, you can't change the value of the double.  Because `value_ptr` is a pointer, you have the variable it points to.  And you can change the value of the variable without using `set_value()`.  transmissions are using it that way.
- `double * const value_ptr` means const pointer to double. This might work since we only set `value_ptr` in the constructors.  But because `value_ptr` is `protected` instead of `private`, derived classes are free to change it.

my two cents.

mhubii, 2024-09-12:
you are perfectly right @yashi. Just asking because lets assume a hardware manufacturer exposes some system state in their SDK a la

```cpp
class Robot {
public:
    const state_t& get_state() const { return _state; };
}
```

in that case (to my best knowledge), the compiler cannot guarantee const correctness and `ReadOnlyHandle` cannot be implemented.

closed 2025-05-19 by github-actions[bot]