ros-controls/ros2_control #1200: spawner does not work with discovery-server

Timple, 2023-12-11:
**Describe the bug**
The [fastdds discovery server](https://docs.ros.org/en/foxy/Tutorials/Advanced/Discovery-Server/Discovery-Server.html) is optimized in the sense that one cannot (by default) see all nodes/topics/services.

The spawner therefor observes wrongfully that the controller_manager is not present and waits until a timeout occurs.

**To Reproduce**
Steps to reproduce the behavior:
1. Terminal 1: `fastdds discovery -i 0 -l localhost`
2. Terminal 2: `export ROS_DISCOVERY_SERVER=localhost:11811 && ros2 launch ros2_control_demo_example_2 diffbot.launch.py`

**Expected behavior**
Everything works the same as without discovery server.

**Screenshots**
Not helpful

**Environment (please complete the following information):**
- Version: Iron

**Proposed solution**
I think crawling nodes and services to find your other half is an anti-pattern. 

We could use a pythonic "Ask forgiveness not permission" method. Try to call the service until is succeeds or a timeout is reached. 

Can you agree on such an implementation? Than we could implement this.

fmauch, 2023-12-26:
I was just about to start implementing a fix for #1182 when stumbling upon this. I would be nice to fix both issues with a change.

@Timple crawling for nodes is one thing. From my understanding waiting for services is not an anti-pattern but [the usual way](https://docs.ros.org/en/rolling/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Service-And-Client.html#write-the-client-node) to make sure your target service is actually there before calling it. 

Removing the `wait_for_controller_manager` functionality makes use of [`wait_for_service`](https://github.com/ros-controls/ros2_control/blob/e92b6c2c84946c8dab44146a5b4a666390543d61/controller_manager/controller_manager/controller_manager_services.py#L38) internally which seems to work fine with the discovery server. If this also solves my use case from #1182 I'll provide a PR implementing this.

destogl, 2024-08-15:
resolved in #1562

closed 2024-08-15 by destogl