ros-controls/ros2_control #1235: Improve robustness of transmission tests

bmagyar, 2023-12-18:
## Background

Improve the robustness of transmission tests by making sure all floating-point value comparisons are using an approximate equality check.

In practice, this means that with Google Mock we use ASSERT_NEAR instead of ASSERT_EQ

Here's an example error we see on the CI from time-to-time:
```
 build/transmission_interface/test_results/transmission_interface/test_four_bar_linkage_transmission.gtest.xml: 8 tests, 0 errors, 1 failure, 0 skipped
- transmission_interface.BlackBoxTest IdentityMap
  <<< failure message
    /home/runner/work/ros2_control/ros2_control/.work/target_ws/src/ros2_control/transmission_interface/test/four_bar_linkage_transmission_test.cpp:174
    Value of: ref_val[1]
    Expected: is approximately 508.18968922836888 (absolute error <= 9.9999999999999995e-07)
      Actual: 508.19 (of type double), which is -1.0207e-06 from 508.19
    /home/runner/work/ros2_control/ros2_control/.work/target_ws/src/ros2_control/transmission_interface/test/four_bar_linkage_transmission_test.cpp:174
    Value of: ref_val[1]
    Expected: is approximately 177.99605718539684 (absolute error <= 9.9999999999999995e-07)
      Actual: 177.996 (of type double), which is 1.38017e-06 from 177.996
```

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/rolling/Tutorials/Workspace/Creating-A-Workspace.html), for Step 3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/rolling/Tutorials.html)
* [Robotics Stack Exchange](https://robotics.stackexchange.com)

**Good luck with your first issue!**

taDachs, 2023-12-19:
I would like to pick up this issue.

christophfroehlich, 2023-12-22:
I think this was solved by #1238? Thanks @taDachs

closed 2023-12-22 by christophfroehlich