ros-controls/ros2_control #1261: ROS 2 Control Node as Composable

mhubii, 2023-12-29:
From the `CMake` it appears the  `ros2_control_node` is compiled into an executable, refer

https://github.com/ros-controls/ros2_control/blob/54ced2a7be75750ca2cd00a7e8d572b65095bc4c/controller_manager/CMakeLists.txt#L42C29-L42C29

Are there plans to provide composition in the future?

mateusmenezes95, 2024-01-20:
The ros2_control offers the ros2_control_node as start point and user friendly to run the ros2_control stuffs. However, nothing stops you from creating your own node. Take a look in the [ros2_control_node.cpp](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/src/ros2_control_node.cpp). You can embed the same logic in a composition node. Just make sure you create the controller manager object and the control loop thread.

mhubii, 2024-01-20:
Thank you for coming back. You are 100% correct @mateusmenezes95. 

The question is more whether this should be provided, and why it is not.

christophfroehlich, 2024-01-20:
I think this would be a minimal change and could be useful. E.g., just used the robot_state_publisher in a manual composition. 
Would you like to create a PR?

mhubii, 2024-01-21:
thank you for coming back @christophfroehlich . Yes, I will have a look at that tomorrow/ early next week, unless someone jumps ahead

mhubii, 2024-03-01:
Duplicate of #330 ?

closed 2025-04-02 by mhubii