ros-controls/ros2_control #1267: [CI] Check duration of controller_managers_srv test

christophfroehlich, 2024-01-03:
> Do you think it make sense to relax the test by reducing the number of controllers? or we simply increase the timeout? Any opinions on this are welcome. If we decided to reduce the number, I can open another PR reducing the number

If the test makes sense this shouldn't be a problem. I'm just curious why it takes so long actually? There is no sleep or timing in there? Does the service call take so much time?

_Originally posted by @christophfroehlich in https://github.com/ros-controls/ros2_control/issues/1260#issuecomment-1872180783_

saikishor, 2024-01-05:
Hello!

I just looked into this issue. It seems like, this is happening due to the the limit of the CPU cores. On my PC, the test runs for 33 seconds
<details>

```{r}
6: [ RUN      ] TestControllerManagerSrvs.list_large_number_of_controllers_with_chains
6: [WARN] [1704444119.165846149] [test_controller_manager]: 'update_rate' parameter not set, using default value.
6: [INFO] [1704444119.166026852] [test_controller_manager]: Subscribing to '/test_controller_manager/robot_description' topic for robot description.
6: [INFO] [1704444119.167658860] [test_controller_manager]: Received robot description from topic.
6: [INFO] [1704444119.167782832] [resource_manager]: Loading hardware 'TestActuatorHardware' 
6: [INFO] [1704444119.167900694] [resource_manager]: Initialize hardware 'TestActuatorHardware' 
6: [INFO] [1704444119.167918705] [resource_manager]: Successful initialization of hardware 'TestActuatorHardware'
6: [INFO] [1704444119.167950555] [resource_manager]: Loading hardware 'TestSensorHardware' 
6: [INFO] [1704444119.168033497] [resource_manager]: Initialize hardware 'TestSensorHardware' 
6: [INFO] [1704444119.168045967] [resource_manager]: Successful initialization of hardware 'TestSensorHardware'
6: [INFO] [1704444119.168057277] [resource_manager]: Loading hardware 'TestSystemHardware' 
6: [INFO] [1704444119.168135668] [resource_manager]: Initialize hardware 'TestSystemHardware' 
6: [INFO] [1704444119.168149509] [resource_manager]: Successful initialization of hardware 'TestSystemHardware'
6: [INFO] [1704444119.168220110] [resource_manager]: 'configure' hardware 'TestSystemHardware' 
6: [INFO] [1704444119.168252270] [resource_manager]: Successful 'configure' of hardware 'TestSystemHardware'
6: [INFO] [1704444119.168261430] [resource_manager]: 'activate' hardware 'TestSystemHardware' 
6: [INFO] [1704444119.168264940] [resource_manager]: Successful 'activate' of hardware 'TestSystemHardware'
6: [INFO] [1704444119.168272241] [resource_manager]: 'configure' hardware 'TestSensorHardware' 
6: [INFO] [1704444119.168276051] [resource_manager]: Successful 'configure' of hardware 'TestSensorHardware'
6: [INFO] [1704444119.168280331] [resource_manager]: 'activate' hardware 'TestSensorHardware' 
6: [INFO] [1704444119.168283751] [resource_manager]: Successful 'activate' of hardware 'TestSensorHardware'
6: [INFO] [1704444119.168289281] [resource_manager]: 'configure' hardware 'TestActuatorHardware' 
6: [INFO] [1704444119.168293091] [resource_manager]: Successful 'configure' of hardware 'TestActuatorHardware'
6: [INFO] [1704444119.168298961] [resource_manager]: 'activate' hardware 'TestActuatorHardware' 
6: [INFO] [1704444119.168302061] [resource_manager]: Successful 'activate' of hardware 'TestActuatorHardware'
6: [INFO] [1704444119.220322652] [srv_client]: Initializing controller : test_random_controllers_0 !
6: [INFO] [1704444119.220343073] [srv_client]: Initializing controller : test_random_controllers_1 !
6: [INFO] [1704444119.220350133] [srv_client]: Initializing controller : test_random_controllers_2 !
6: [INFO] [1704444119.220356263] [srv_client]: Initializing controller : test_random_controllers_3 !
6: [INFO] [1704444119.220362363] [srv_client]: Initializing controller : test_random_controllers_4 !
6: [INFO] [1704444119.220368393] [srv_client]: Initializing controller : test_random_controllers_5 !
6: [INFO] [1704444119.220374453] [srv_client]: Initializing controller : test_random_controllers_6 !
6: [INFO] [1704444119.220380443] [srv_client]: Initializing controller : test_random_controllers_7 !
6: [INFO] [1704444119.220386553] [srv_client]: Initializing controller : test_random_controllers_8 !
6: [INFO] [1704444119.220392514] [srv_client]: Initializing controller : test_random_controllers_9 !
6: [INFO] [1704444119.220398514] [srv_client]: Initializing controller : test_random_controllers_10 !
6: [INFO] [1704444119.220405744] [srv_client]: Initializing controller : test_random_controllers_11 !
6: [INFO] [1704444119.220411814] [srv_client]: Initializing controller : test_random_controllers_12 !
6: [INFO] [1704444119.220417904] [srv_client]: Initializing controller : test_random_controllers_13 !
6: [INFO] [1704444119.220424024] [srv_client]: Initializing controller : test_random_controllers_14 !
6: [INFO] [1704444119.220430134] [srv_client]: Initializing controller : test_random_controllers_15 !
6: [INFO] [1704444119.220436144] [srv_client]: Initializing controller : test_random_controllers_16 !
6: [INFO] [1704444119.220442254] [srv_client]: Initializing controller : test_random_controllers_17 !
6: [INFO] [1704444119.220448415] [srv_client]: Initializing controller : test_random_controllers_18 !
6: [INFO] [1704444119.220454735] [srv_client]: Initializing controller : test_random_controllers_19 !
6: [INFO] [1704444119.220460775] [srv_client]: Initializing controller : test_random_controllers_20 !
6: [INFO] [1704444119.220466815] [srv_client]: Initializing controller : test_random_controllers_21 !
6: [INFO] [1704444119.220472885] [srv_client]: Initializing controller : test_random_controllers_22 !
6: [INFO] [1704444119.220479185] [srv_client]: Initializing controller : test_random_controllers_23 !
6: [INFO] [1704444119.220485385] [srv_client]: Initializing controller : test_random_controllers_24 !
6: [INFO] [1704444119.220491735] [srv_client]: Initializing controller : test_random_controllers_25 !
6: [INFO] [1704444119.220497825] [srv_client]: Initializing controller : test_random_controllers_26 !
[... cut ...]
[INFO] [1703851858.021174567] [test_controller_manager]: Configuring controller 'test_broadcaster_80'
>>>
build/controller_manager/test_results/controller_manager/test_controller_manager_srvs.gtest.xml: 1 test, 1 error, 0 failures, 0 skipped
- controller_manager test_controller_manager_srvs.gtest.missing_result
<<< error message
The test did not generate a result file.
>>>
```

</details>

As this is the case, I recommend reducing the number of controllers for the sake for determinism and less run time on Ci

saikishor, 2024-01-05:
I've also tried to reduce the number of CPUs used by my docker container recommended by as @moriarty. I also see that the runtime of the tests is nearly double.

christophfroehlich, 2024-01-07:
Can't we improve the speed of the service itself? Or which part is taking such a long time per controller?

christophfroehlich, 2024-10-10:
@saikishor I analyzed the times for the failing CI in https://github.com/ros-controls/ros2_control/issues/1700 with https://github.com/ros-controls/ros2_control/pull/1781

Times are in ms, iron behaves similar to rolling.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/FROEHL~1/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/FROEHL~1/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#0563C1" vlink="#954F72">

  | humble | iron
-- | -- | --
TestControllerManagerSrvs.list_controller_types | 107 | 123
TestControllerManagerSrvs.list_controllers_srv | 192 | 209
TestControllerManagerSrvs.list_chained_controllers_srv | 166 | 192
TestControllerManagerSrvs.reload_controller_libraries_srv | 439 | 446
TestControllerManagerSrvs.load_controller_srv | 136 | 145
TestControllerManagerSrvs.unload_controller_srv | 145 | 158
TestControllerManagerSrvs.configure_controller_srv | 194 | 209
TestControllerManagerSrvs.list_sorted_chained_controllers | 217 | 240
TestControllerManagerSrvs.list_sorted_complex_chained_controllers | 263 | 285
TestControllerManagerSrvs.list_sorted_independent_chained_controllers | 312 | 339
TestControllerManagerSrvs.list_large_number_of_controllers_with_chains | 185696 | 31100
TestControllerManagerSrvs.list_sorted_large_chained_controller_tree | 22836 | 6420
TestControllerManagerSrvs.activate_chained_controllers_one_by_one | 217 | 225
TestControllerManagerSrvs.activate_chained_controllers_all_at_once | 171 | 190

</body>

</html>

#1384 hasn't been backported to iron, why is iron fine but humble not? 
How should we fix this in humble:

- increase timeout
- decrease amount of controllers in the tests
- anything else?

saikishor, 2024-10-10:
@christophfroehlich if the Iron doesn't have #1384 it doesn't make a lot of sense as you said. Moreover, it's like 6 times. Maybe we can reduce the number for now and see. Moreover, in that test, we have like 100+ controllers, this is not so usual case, so it shouldn't be a lot of problem.

What do you say?

christophfroehlich, 2024-10-10:
I have no idea actually :D You are more familiar with this test, could you have a quick look and compare the test from humble/iron? 
If you don't find a possible conceptional cause of this, I prefer to reduce the number of controllers for the tests on humble to get back to "stable" CI.

saikishor, 2024-10-10:
Sure, I will check it once I get some free time :)

christophfroehlich, 2025-12-31:
#1700
@saikishor this still fails everyday on humble source build. Even with 240s timeout, see #1781 
Can we just remove the numbers of controllers, or find out why it take so long on humble?