ros-controls/ros2_control #1277: URDF <gpio> component state interface "initial_value" param not getting read.

Nibanovic, 2024-01-08:
**Describe the bug**
I've been trying to set up `scaled_joint_trajectory_controller` for my manipulator, like in [UR_robot_driver](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/humble), and test it using mocked hardware.
To do that, I need a `<gpio>` component which will hold the `scaling_factor` value.

Example component:
```
<gpio name="speed_scaling">
  <state_interface name="speed_scaling_factor">
    <param name="initial_value">1.0</param>
  </state_interface>
</gpio>
```

When setting the `initial_value` param, like in `<joint>` components for example, the param is not getting read.

**To Reproduce**
1. Add the above component to your `<ros2_control>` tag in URDF
2. Use `mock_components/GenericSystem` as hardware
3. Add the `speed_scaling_state_broadcaster` controller from [ur_controllers](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/humble/ur_controllers/src/speed_scaling_state_broadcaster.cpp), which publishes the state interface of the component
4. Launch the `ros2_control` node, `robot_state_publisher` node and the `speed_scaling_state_broadcaster`
5. `ros2 topic echo /speed_scaling_state_broadcaster/speed_scaling`

**Expected behavior**
I would expect the topic to echo `100.0`, as `speed_scaling_state_broadcaster` outputs the factor as %.
The actual output is `.nan`.

**More Context**
If I change the `<gpio>` tag to `<joint>`, so that the component looks like this:
```
<joint name="speed_scaling">
  <state_interface name="speed_scaling_factor">
    <param name="initial_value">1.0</param>
  </state_interface>
</joint>
```
... the component behaves as expected with mocked hardware. This leads me to believe that there could be an issue with the way `<gpio>` param `initial value` is parsed in the `mock_components/GenericSystem`

**Environment:**
 - OS: Ubuntu
 - Version: Humble -> both `ros2_control` and `ur_controllers` packages are `humble`

christophfroehlich, 2024-01-11:
I was not able to reproduce your issue with https://github.com/ros-controls/ros2_control_demos/pull/429, neither on rolling or humble. Can you have a look at this PR, if there is a difference to your configuration?

Btw there is already a test for that
https://github.com/ros-controls/ros2_control/blob/71506d5f12e1e8529be4951ccf45f7144969f841/hardware_interface/test/mock_components/test_generic_system.cpp#L1662-L1679

bailaC, 2024-01-12:
@Nibanovic , will you be able to check this behavior with rolling? I see some code changes around the mock components in rolling. Just asking to verify this behavior on rolling.

bailaC, 2024-01-13:
Thank you @christophfroehlich  for enhancing example 10. Your changes are working well for me for rolling. Humble yet to be checked.

bailaC, 2024-01-14:
Tried your test @christophfroehlich  with humble also. Working as expected.

Nibanovic, 2024-01-15:
I have rebuilt my packages and retested everything as described in the issue. Funnily enough, now it is working as expected. Thanks for your time, apologies for the non-issue.

closed 2024-01-15 by Nibanovic