ros-controls/ros2_control #1285: ros2_controller CI is broken for Iron

christophfroehlich, 2024-01-12:
See the following jobs: 

- https://github.com/ros-controls/ros2_controllers/actions/runs/7436961863/job/20403754972
- https://github.com/ros-controls/ros2_controllers/actions/runs/7436961900/job/20234036118

For humble or rolling the same tests work.
It seems that test_components are not exported; they are in the BUILD_TESTING branch. 

https://github.com/ros-controls/ros2_control/blob/71506d5f12e1e8529be4951ccf45f7144969f841/hardware_interface/CMakeLists.txt#L82-L96

@bmagyar Is it possible that iron packages are built with BUILD_TESTING false, but others with true?

saikishor, 2024-01-12:
Thanks for bringing it up!

At PAL, we also have a similar issue with the `test_components`, when built on our internal CI. It doesn't build this library, hence for all the internally developed controllers, we couldn't add similar tests. Atleast, we build our debians with the `BUILD_TESTING` flag to `false`.

destogl, 2024-01-15:
It might be that because `ros2_control_test_assets` is only dependent on the tests, it is not shipped or appropriately added to the hierarchy. I believe that we use it only in the testing part as a dependency.

christophfroehlich, 2024-01-17:
Should we change it and export it for `BUILD_TESTING` false also? or can we change the flag on the build farm to match the other distros?

saikishor, 2024-01-17:
I think it is better to place it outside the tests and export it so that the other controller tests can use it. All CIs will more or less have the similar issue as it is not normal to generate your debians with `BUILD_TESTING` flag set to `true`.

christophfroehlich, 2024-01-17:
conclusion of today's meeting: Lets make `hardware_interface_test` package, which installs the test components and move the tests from `hardware_interface` using those components also there to break the circular dependency.

PR #1325 (2024-01-25): Move `test_components` to own package
Move `test_components` to their own package, together with tests of hardware_interface package relying on it to break circular dependency.

We decided this in the WG meeting 2024 01 17

This will fix #1285 after we change the test-dependency of the ros2_controllers.
closed 2024-01-25 by bmagyar