ros-controls/ros2_control #1297: Raspberry Pi and Pipgio

Russ76, 2024-01-15:
**Is your feature request related to a problem? Please describe.**
I'm trying to integrate this controller with the Pigpio module on Raspberry Pi 4b. This module is very fast and reliable. My robot is a differential drive robot, needing just the two controllers. Pigpio is written in C. I'm having trouble with passing the variables, proper classes, and making all work together. I have Pigpio configured now to write encoder positions to small text files on a ramdisk. The controller reads these. This seems to be working but doesn't seem ideal. The write routine gives even more trouble. My motor controller (Sabertooth 2x25) is easy to control with a serial interface, by sending just an integer between 0 and 256 to the Pigpio write_serial pin. If I could get this to work it would be very fast and precise. I am using the Example_2 Diffbot code.

**Describe the solution you'd like**
This tight integration of Ros2_Control and Pigpio could be useful to many roboticists. 

**Describe alternatives you've considered**
I developed Python modules for the IMU, encoders and motor control that worked well with Pigpio on the Raspberry Pi using Ros Humble, without Ros2_Control. An advantage of this setup was the second WiFi channel that Raspi allows, so the main robot computer can run the Lidar, camera and mapping modules, which require substantial WiFi bandwidth. It worked well for mapping. But for autonomous driving, it was jerky and unreliable. It lacked the tight integration of movement control that Ros2_control offers.

**Additional context**
Add any other context or screenshots about the feature request here.

mateusmenezes95, 2024-01-20:
Hello @Russ76, like seems that your problem may be solved by creating your own Hardware Component because controllers under the ros2_control concepts does not communicate directly with the hardware. In addition, the goal of the ros2_control is to provide a framework for developers integrate it in their own robots. The package [ros_control_demos](https://github.com/ros-controls/ros2_control_demos/tree/master) has some (many) examples of how to create hardware components. You can also take a look in the [Writing a Hardware Component](https://control.ros.org/master/doc/ros2_control/hardware_interface/doc/writing_new_hardware_component.html#writing-a-hardware-component) tutorial and [this video](https://www.youtube.com/watch?v=J02jEKawE5U&t=1020s)

Russ76, 2024-01-20:
Thanks, I have looked at these resources. Unfortunately, I'm not an expert at C++ programming, and am having trouble getting anything going. I'm confused by what variables I need to connect with, and whether they are vectors or simple variables, and whether they will be retained between functions or whether they should be static or belong to a certain class, etc. Also, Pigpio is written in C, not C++, so this creates another difficulty. I was able to get my own modules working in Python, but Ros2_control doesn't work with Python, unfortunately.

Russ76, 2024-01-26:
We would like to hire a Ros2_control expert to write this module for us. Does anyone want another job?

closed 2025-05-19 by github-actions[bot]