ros-controls/ros2_control #1299: [Humble] Gazebo simulation failing after merging #1272

Noel215, 2024-01-16:
Hi,

After merging the PR https://github.com/ros-controls/ros2_control/pull/1272 the TIAGo simulation is failing. The original issue is reported in https://github.com/pal-robotics/tiago_simulation/issues/28.
Could you take a look, please?

I've been running as well the `diff_drive` example in the `gazebo_ros2_control_demos`:
```
ros2 launch gazebo_ros2_control_demos diff_drive.launch.py
```
and its output is the following
```
[gzserver-1] [INFO] [1705401148.489354723] [resource_manager]: Initialize hardware 'GazeboSystem' 
[gzserver-1] [INFO] [1705401148.489951706] [resource_manager]: Successful initialization of hardware 'GazeboSystem'
[gzserver-1] [INFO] [1705401148.491561292] [resource_manager]: 'configure' hardware 'GazeboSystem' 
[gzserver-1] [INFO] [1705401148.491615401] [resource_manager]: Successful 'configure' of hardware 'GazeboSystem'
[gzserver-1] [INFO] [1705401148.491673808] [resource_manager]: 'activate' hardware 'GazeboSystem' 
[gzserver-1] [INFO] [1705401148.491677966] [resource_manager]: Successful 'activate' of hardware 'GazeboSystem'
[gzserver-1] [INFO] [1705401148.491962439] [gazebo_ros2_control]: Loading controller_manager
[gzserver-1] [INFO] [1705401148.506958281] [controller_manager]: Subscribing to '~/robot_description' topic for robot description file.
[gzserver-1] [WARN] [1705401148.511946567] [gazebo_ros2_control]:  Desired controller update period (0.01 s) is slower than the gazebo simulation period (0.001 s).
[gzserver-1] [INFO] [1705401148.512763337] [gazebo_ros2_control]: Loaded gazebo_ros2_control.
[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 119154]
[INFO] [ros2-5]: process started with pid [119278]
[ros2-5] Could not contact service /controller_manager/load_controller
[ERROR] [ros2-5]: process has died [pid 119278, exit code 1, cmd 'ros2 control load_controller --set-state active joint_state_broadcaster'].
[INFO] [ros2-6]: process started with pid [119438]
[ros2-6] Could not contact service /controller_manager/load_controller
[ERROR] [ros2-6]: process has died [pid 119438, exit code 1, cmd 'ros2 control load_controller --set-state active diff_drive_base_controller'].
```

I think something of that PR has broken the `GazeboSystem` plugin in https://github.com/ros-controls/gazebo_ros2_control/.

saikishor, 2024-01-16:
The issue is coming from the `resource_manager` object created in the gazebo_ros2_control_plugin, as it is not initialized with the URDF. With the recent PR that we merged regarding the initializing of CM services only after the resource manager is completely initialized, this is breaking. We need to find a proper solution to fix this issue.

https://github.com/ros-controls/gazebo_ros2_control/blob/60e4b75f4a786c9563d7dd97c6d08a31ca69c5b1/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp#L365-L371

christophfroehlich, 2024-01-16:
I also ran into the same problem, reproducible also with `ros2 launch ros2_control_demo_example_9 rrbot_gazebo_classic.launch.py`

should we revert #1272 until we've found a solution?

saikishor, 2024-01-16:
I think reverting #1272 will bring back the same problem of: https://github.com/ros-controls/ros2_control/issues/1262. I'll try to find some solution soon. The solution seems to add a parameter to the `load_urdf` method inside the resource manager, not to load and initialize the system or somehow get the URDF into the resource manager

commit 510ba730 (2024-01-17): [RM] Fix crash for missing urdf in resource manager (#1301)

* check the components size instead of the variable (fixes #1299)
* add new load_and_initialize_components default argument to the load_urdf method
commit d8cb7b9d (2024-01-18): [RM] Fix crash for missing urdf in resource manager (#1301)

* check the components size instead of the variable (fixes #1299)
* add new load_and_initialize_components default argument to the load_urdf method

(cherry picked from commit 510ba7309921339a3b4e40835efc5a7fcddb225f)
closed 2024-01-17 by destogl