ros-controls/ros2_control #1314: Error while trying to create a custom package using ros2_control (Windows)

GilmarCorreia, 2024-01-17:
**Describe the bug**
Hello ROS Developers,

Today I encountered a new issue while attempting to build a custom ROS2 package (sim_ros2_control) that utilizes some ROS2 control packages (hardware_interfaces, controller_manager, ...). I have successfully built the same package on Linux 22.04, but now I'm trying to build it on Windows (with all the necessary Visual Studio compilers and dependencies).

When I attempt to run colcon build, the following errors appear (excerpt from the log):

```txt
C:\Users\user\Desktop\ros2_ws\install\hardware_interface\include\hardware_interface/types/hardware_interface_return_values.hpp(25,3): error C2143: syntax error: missing '}' before 'constant' (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\Users\user\Desktop\ros2_ws\install\hardware_interface\include\hardware_interface/types/hardware_interface_return_values.hpp(25,3): error C2059: syntax error: 'constant' (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\Users\user\Desktop\ros2_ws\install\hardware_interface\include\hardware_interface/types/hardware_interface_return_values.hpp(26,1): error C2143: syntax error: missing ';' before '}' (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\Users\user\Desktop\ros2_ws\install\hardware_interface\include\hardware_interface/types/hardware_interface_return_values.hpp(28,1): error C2059: syntax error: '}' (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\Users\user\Desktop\ros2_ws\install\hardware_interface\include\hardware_interface/types/hardware_interface_return_values.hpp(28,1): error C2143: syntax error: missing ';' before '}' (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\Users\user\Desktop\ros2_ws\install\hardware_interface\include\hardware_interface/resource_manager.hpp(33,1): error C2143: syntax error: missing ';' before '{' (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\Users\user\Desktop\ros2_ws\install\hardware_interface\include\hardware_interface/resource_manager.hpp(33,1): error C2447: '{': missing function header (old-style formal list?) (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\ros2\humble\install\include\rclcpp_lifecycle\rclcpp_lifecycle/node_interfaces/lifecycle_node_interface.hpp(53,5): error C2143: syntax error: missing '}' before 'constant' (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\ros2\humble\install\include\rclcpp_lifecycle\rclcpp_lifecycle/node_interfaces/lifecycle_node_interface.hpp(53,5): error C2059: syntax error: 'constant' (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\ros2\humble\install\include\rclcpp_lifecycle\rclcpp_lifecycle/node_interfaces/lifecycle_node_interface.hpp(54,3): error C2143: syntax error: missing ';' before '}' (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\ros2\humble\install\include\rclcpp_lifecycle\rclcpp_lifecycle/node_interfaces/lifecycle_node_interface.hpp(54,3): error C2238: unexpected token(s) preceding ';' (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\ros2\humble\install\include\rclcpp_lifecycle\rclcpp_lifecycle/node_interfaces/lifecycle_node_interface.hpp(62,15): error C2433: 'CallbackReturn': 'virtual' not permitted on data declarations (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\ros2\humble\install\include\rclcpp_lifecycle\rclcpp_lifecycle/node_interfaces/lifecycle_node_interface.hpp(62,15): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\ros2\humble\install\include\rclcpp_lifecycle\rclcpp_lifecycle/node_interfaces/lifecycle_node_interface.hpp(62,3): error C2146: syntax error: missing ';' before identifier 'on_configure' (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\ros2\humble\install\include\rclcpp_lifecycle\rclcpp_lifecycle/node_interfaces/lifecycle_node_interface.hpp(62,45): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\ros2\humble\install\include\rclcpp_lifecycle\rclcpp_lifecycle/node_interfaces/lifecycle_node_interface.hpp(70,13): error C2433: 'CallbackReturn': 'virtual' not permitted on data declarations (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\ros2\humble\install\include\rclcpp_lifecycle\rclcpp_lifecycle/node_interfaces/lifecycle_node_interface.hpp(70,13): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\ros2\humble\install\include\rclcpp_lifecycle\rclcpp_lifecycle/node_interfaces/lifecycle_node_interface.hpp(70,3): error C2146: syntax error: missing ';' before identifier 'on_cleanup' (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\ros2\humble\install\include\rclcpp_lifecycle\rclcpp_lifecycle/node_interfaces/lifecycle_node_interface.hpp(70,43): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\ros2\humble\install\include\rclcpp_lifecycle\rclcpp_lifecycle/node_interfaces/lifecycle_node_interface.hpp(78,14): error C2433: 'CallbackReturn': 'virtual' not permitted on data declarations (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\ros2\humble\install\include\rclcpp_lifecycle\rclcpp_lifecycle/node_interfaces/lifecycle_node_interface.hpp(78,14): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
C:\ros2\humble\install\include\rclcpp_lifecycle\rclcpp_lifecycle/node_interfaces/lifecycle_node_interface.hpp(78,3): error C2146: syntax error: missing ';' before identifier 'on_shutdown' (compiling source file C:\Users\use [... cut ...]
C:\ros2\humble\install\include\rclcpp_lifecycle\rclcpp_lifecycle/lifecycle_node.hpp(789,36): error C2653: 'LifecycleNodeInterface': is not a class or namespace name (compiling source file C:\Users\user\Desktop\ros2_ws\src\sim_ros2_control\src\sim_ros2_control_plugin.cpp) [C:\Users\user\Desktop\ros2_ws\build\sim_ros2_control\simROS2Control.vcxproj]
...
```

It appears that I may not be using the correct compiler, but I'm currently out of ideas on how to resolve these errors while building on Windows. Has anyone else encountered a similar issue?

**Environment:**
 - OS: Windows 10
 - Version: Humble

christophfroehlich, 2024-01-21:
None of the core contributors use ROS 2/ ros2_control on windows afaik, that's why I don't expect we can give proper help here. Maybe you can find more people using ROS 2 on windows on https://robotics.stackexchange.com?

But please report here if you have solved your issue. We also would appreciate if you could provide installation instructions to our documentation if additional steps to https://docs.ros.org/en/rolling/Installation/Windows-Install-Binary.html are necessary. I could also think of a windows CI pipeline, it seems that github provides windows runners now. Contributions are welcome!

christophfroehlich, 2024-07-19:
May be solved with https://github.com/ros-controls/ros2_controllers/issues/1053

GilmarCorreia, 2024-07-22:
Thank you @christophfroehlich, I will test it!