ros-controls/ros2_control #1400: Cannot activate multiple chainable controllers at once anymore

AugusteBourgois, 2024-02-16:
Hi everyone,

We have met the following bug since the last update of ros2_control (2.37.0).

So we basically have three controllers C1, C2 and C3. C1 and C2 are chainable controllersn while C3 is a basic controller.
C1 claims hardware interfaces and exposes chainable interfaces, in turn claimed by C2 which exposes its own chainable interfaces, at last claimed by C3.

Before the update, we were able to `ros2 control switch_controllers --activate C1 C2 C3`.

After the last update, this very same command fails, the error message stating that the chainable interfaces exposed by C1 and C2 are not available. This feels like a feature loss, as this means we need to activate controllers one by one.

**To Reproduce**
I have implemented a minimal 'not' working example in the form of two unit tests for a controller manager, please refer to [https://github.com/forssea-robotics/ros2_control/blob/006377d20985f832e3694c64eaae7952e738db43/controller_manager/test/test_controller_manager_srvs.cpp#L1523](https://github.com/forssea-robotics/ros2_control/blob/006377d20985f832e3694c64eaae7952e738db43/controller_manager/test/test_controller_manager_srvs.cpp#L1523)

The first one activate controllers one by one and passes, while the second one activates them all at once and fails with this message:

```
[ERROR 1708098723.350215573] [resource_manager] [prepare_command_mode_switch() at /home/abourgois/dev/ros2_control_ws/src/ros2_control/hardware_interface/src/resource_manager.cpp:1070]:
Not acceptable command interfaces combination: 
Start interfaces: 
[
  test_chainable_controller_name_2/joint1/position
  test_chainable_controller_name_2/joint1/velocity
  joint2/velocity
  test_chainable_controller_name_1/joint1/position
  joint1/position
]
Stop interfaces: 
[
]
Not available: 
[
 test_chainable_controller_name_2/joint1/position
 test_chainable_controller_name_2/joint1/velocity
 test_chainable_controller_name_1/joint1/position
]

[ERROR 1708098723.350243418] [test_controller_manager] [switch_controller() at /home/abourgois/dev/ros2_control_ws/src/ros2_control/controller_manager/src/controller_manager.cpp:1104]:
Could not switch controllers since prepare command mode switch was rejected.
/home/abourgois/dev/ros2_control_ws/src/ros2_control/controller_manager/test/test_controller_manager_srvs.cpp:229: Failure
Expected equality of these values:
  res
    Which is: 1-byte object <01>
  controller_interface::return_type::OK
    Which is: 1-byte object <00>
[  FAILED  ] TestControllerManagerSrvs.activate_chained_controllers_all_at_once (128 ms)
```

**Expected behavior**
`ros2 control switch_controllers --activate C1 C2 C3` should activate all three controllers as before, and my second unit test should pass.

**Environment**
 - OS: Ubuntu 22.04
 - Version Humble, ros2_control 2.37.0

saikishor, 2024-02-16:
Hello @AugusteBourgois!

Thank you for the nice bug description and the proper test case. I was able to reproduce the same. I've just fixed it in the PR #1401. The tests seem to pass now.

Thank you,

Best Regards,
Sai

AugusteBourgois, 2024-02-19:
@saikishor ,

I confirm that you fix works for us once backported to humble.

Hopefully this will get released soon (cherry picks were easily done for us).

Thank you again for your reactivity,
Kindly
Auguste

saikishor, 2024-02-19:
@AugusteBourgois 
Thanks for confirming it. Can you please approve the PR #1401. Let's hope it's done for the following sync.

Thank you!

closed 2024-02-22 by bmagyar