ros-controls/ros2_control #1416: why my ros2 control has no fun“joint_limits_urdf.hpp” in package joint_limits

PANJIACHENG-98, 2024-02-26:
i use cli sudo apt install ros-humble-ros2-control* to install the ros2 control,but has no fun "joint_limits_urdf.hpp” in package joint_limit ,why ??????
 i check source package in https://github.com/ros-controls/ros2_control,in branch master , the package joint_limit contains 3 function include "joint_limits_urdf.hpp”,but in humble branch ,only 2 functions,without "joint_limits_urdf.hpp”,
pllease tell me why ??? and use sudo apt install ros-humble-ros2-control* to install which branch ?
[image]
[image]

christophfroehlich, 2024-02-26:
This was added recently on rolling https://github.com/ros-controls/ros2_control/pull/1298 and not backported to humble/iron.
If you need it, we maybe can backport it to humble

PANJIACHENG-98, 2024-02-26:
yes yes i really need it ,please backport it!!

christophfroehlich, 2024-03-04:
It was decided to not backport it but we are working on making the rolling branches compileable on humble+rolling. In the meantime you can just simply download the file and include it in your project (it is not used elsewhere in the code yet).

closed 2024-03-04 by christophfroehlich