ros-controls/ros2_control #1450: Controller requiring robot_description parameter does not load anymore after switch to loading robot_description from topc

firesurfer, 2024-03-19:
**Describe the bug**

Please see the related issue in the `cartesian controllers` repository: https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/179

I recently (tried) to switch from passing the `robot_description` as a parameter to loading it from the corresponding topic.
The cartesian controllers apparently depend on obtaining the `robot_description` from the respective parameter. 

This issue is rather a question than a pure bug report: What is the intended way of accessing the robot description inside a controller?

a) Via the robot_description parameter 
b) Load it yourself from the topic
c) A different way (e.g. ros2control provides the URDF ready to use to the controller?)

**To Reproduce**

1. Configure ros2control to load the `robot_description` from the topic
2. Try to load any of the cartesian controllers

**Environment (please complete the following information):**

- I am currently using the fixed branch from @saikishor https://github.com/saikishor/ros2_control/tree/fix/iron/missing/resources_lock

christophfroehlich, 2024-03-19:
The URDF is already part of the base class:
https://github.com/ros-controls/ros2_control/blob/20e01d867fd14c1a3a0b8ea999d41d9d27dcf3bb/controller_interface/include/controller_interface/controller_interface_base.hpp#L253

firesurfer, 2024-03-19:
Ah great to know. Do you also plan on backporting this to iron/humble ?

christophfroehlich, 2024-03-19:
No, I don't think so because #1088 would break API (init is not a method used by users, but still at least breaks ABI for controllers.)

We will stop backporting such changes, and motivate users to build the [latest rolling version for their distros](https://github.com/ros-controls/ros2_control_ci?tab=readme-ov-file#compatibility-versions) from source.

closed 2024-03-19 by christophfroehlich

saikishor, 2024-03-19:
I agree with @christophfroehlich backporting it will break API and ABI. It's recommended from now on to use the rolling version to use the main features