ros-controls/ros2_control #1452: A problem with Controller_manager

AAE1d, 2024-03-20:
Hello
I am fairly new to ros in general and I am trying to implement an Autonomous Robot with LiDAR, I am running on ROS2 Humble, I built the URDF file and the launch files and tried the simulations with the gazebo plugins and it went well and was able to run navigation packages well too. The problem is when I connect the motors and try to control them myself, I have separate launch files and I think that the files are good, The problem that the ros2 control node can't connect to the controller Manager, when I launch the launch file it loads the robot description and then gets the diff_cont spawner to connect to the controller manager, but it can't do that and says "[spawner-3] [INFO] [1709997603.518112113] [spawner_diff_cont]: Waiting for '/controller_manager' services to be available" and stays like that for a bit and times out. This problem was mentioned in some forums but the only solution for it was to remove .py from "spawner.py" in the controller_manager node, I tried it and it didn't work for me.
I know that the motors are fine because I tried them using serial_motor_demo to check if it was working and they worked fine.
Is there any way to solve this?
I'd be happy to share any needed files
Thank you

christophfroehlich, 2024-03-20:
Hi! 
Have you tried our [demos](https://control.ros.org/humble/doc/ros2_control_demos/doc/index.html) and learned how to setup your robot with ros2_control?
If questions remain, please ask your question on https://robotics.stackexchange.com and add more information (your launch file, ros2_control.yaml etc).

closed 2024-03-20 by christophfroehlich

saikishor, 2024-03-20:
Hello @AAE1d!

The controller_manager services are not listed out unless the resource manager is initialized properly, this is done this way to ensure that the spawners don't request for activating controllers even before the hardware is completely setup as in here:

https://github.com/ros-controls/ros2_control/blob/deb34de09a2a961e965c3542d9aaa2d2cf829c84/controller_manager/src/controller_manager.cpp#L281-L293