ros-controls/ros2_control #1453: After  latest update controller manager not working

Ishansehgal, 2024-03-21:
Previously, my control manager was working fine and loading properly, but now that I have updated my packages, it shows waiting for the controller manager and the control manager is not available. I have tried everything and have no clue how to solve this; my whole project depends on it.

TheNoobInventor, 2024-03-21:
Hello Sai, 

I've got the same issue. I'm not sure if this will suffice, but this is my console output:

```
[robot_state_publisher-1] [INFO] [1711055103.211911581] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1711055103.212075991] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1711055103.212096268] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-1] [INFO] [1711055103.212106186] [robot_state_publisher]: got segment camera_link_optical
[robot_state_publisher-1] [INFO] [1711055103.212115730] [robot_state_publisher]: got segment caster_wheel
[robot_state_publisher-1] [INFO] [1711055103.212124489] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-1] [INFO] [1711055103.212132579] [robot_state_publisher]: got segment left_wheel
[robot_state_publisher-1] [INFO] [1711055103.212140344] [robot_state_publisher]: got segment lidar_link
[robot_state_publisher-1] [INFO] [1711055103.212148409] [robot_state_publisher]: got segment right_wheel
[twist_mux-10] [INFO] [1711055103.225610261] [twist_mux]: Topic handler 'topics.joystick' subscribed to topic 'cmd_vel_joy': timeout = 0.500000s , priority = 100.
[twist_mux-10] [INFO] [1711055103.226627108] [twist_mux]: Topic handler 'topics.navigation' subscribed to topic 'cmd_vel': timeout = 0.500000s , priority = 10.
[teleop_node-9] [INFO] [1711055103.230201832] [TeleopTwistJoy]: Teleop enable button 4.
[teleop_node-9] [INFO] [1711055103.230262027] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.400000.
[teleop_node-9] [INFO] [1711055103.230273579] [TeleopTwistJoy]: Angular axis yaw on 2 at scale 1.000000.
[spawn_entity.py-4] [INFO] [1711055103.521518493] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1711055103.521742979] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-4] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-4]   warnings.warn(
[spawn_entity.py-4] [INFO] [1711055103.523543582] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-4] [INFO] [1711055103.534546322] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1711055103.534739546] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1711055103.788348254] [spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-4] [INFO] [1711055103.838006736] [spawn_entity]: Spawn status: Entity pushed to spawn queue, but spawn service timed outwaiting for entity to appear in simulation under the name [lidarbot]
[spawn_entity.py-4] [ERROR] [1711055103.838202663] [spawn_entity]: Spawn service failed. Exiting.
[ERROR] [spawn_entity.py-4]: process has died [pid 68327, exit code 1, cmd '/opt/ros/humble/lib/gazebo_ros/spawn_entity.py -topic robot_description -entity lidarbot --ros-args'].
[gzserver-2] [INFO] [1711055104.714817740] [camera_controller]: Publishing camera info to [/camera/camera_info]
[spawner-5] [INFO] [1711055105.457440918] [spawner_diff_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1711055105.479843301] [spawner_joint_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1711055107.469612840] [spawner_diff_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1711055107.491606951] [spawner_joint_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1711055109.483960882] [spawner_diff_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1711055109.507490946] [spawner_joint_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1711055111.496981392] [spawner_diff_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1711055111.519635922] [spawner_joint_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-5] [ERROR] [1711055113.508282366] [spawner_diff_controller]: Controller manager not available
[spawner-6] [ERROR] [1711055113.530858575] [spawner_joint_broadcaster]: Controller manager not available
[ERROR] [spawner-5]: process has died [pid 68329, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner diff_controller --ros-args'].
[ERROR] [spawner-6]: process has died [pid 68331, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_broadcaster --ros-args'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[teleop_node-9] [INFO] [1711055119.361188669] [rclcpp]: signal_handler(signum=2)
[twist_mux-10] [INFO] [1711055119.361238555] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-1] [INFO] [1711055119.361295838] [rclcpp]: signal_handler(signum=2)
[joy_node-8] [INFO] [1711055119.361252114] [rclcpp]: signal_handler(signum=2)
[ekf_node-7] [INFO] [1711055119.361311079] [rclcpp]: signal_handler(signum=2)
[INFO] [teleop_node-9]: process has finished cleanly [pid 68337]
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 68321]
[INFO] [twist_mux-10]: process has finished cleanly [pid 68339]
[INFO] [ekf_node-7]: process has finished cleanly [pid 68333]
[INFO] [joy_node-8]: process has finished cleanly [pid 68335]
[INFO] [gzserver-2]: process has finished cleanly [pid 68323]
[ERROR] [gzclient-3]: process has died [pid 68325, exit code -11, cmd 'gzclient --ros-args --params-file /home/noobinventor/lidarbot_ws/install/lidarbot_gazebo/share/lidarbot_gazebo/config/gazebo_params.yaml'].
```
I've gone through the ros2_control tutorials to set things up using Gazebo (in the plugin section), made the changes to my URDF files but no success. Please let me know if you need more information, I don't know if the code snippet was what you meant when you asked for errors from the controller manager.

saikishor, 2024-03-21:
Hello @TheNoobInventor,

Can you please let us know the distro and the version of packages you are using?

Thank you

saikishor, 2024-03-21:
Can you test with version 2.39.1 or older version to see if it works?

TheNoobInventor, 2024-03-21:
> Hello @TheNoobInventor,
> 
> Can you please let us know the distro and the version of packages you are using?
> 
> Thank you

I'm currently using ROS2 Humble and the `ros2_control` version is `2.40.0`.

saikishor, 2024-03-21:
> > Hello @TheNoobInventor,
> > 
> > Can you please let us know the distro and the version of packages you are using?
> > 
> > Thank you
> 
> I'm currently using ROS2 Humble and the `ros2_control` version is `2.40.0`.

Can you please clone 2.39.1 into a workspace and test the changes?

TheNoobInventor, 2024-03-21:
I just did that and I still have the same error. I'll try a version lower and see if that changes things.

```
[spawn_entity.py-4]   warnings.warn(
[spawn_entity.py-4] [INFO] [1711059164.507978251] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-4] [INFO] [1711059164.518930442] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1711059164.519135051] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1711059164.522045703] [spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-4] [INFO] [1711059164.544045768] [spawn_entity]: Spawn status: Entity [lidarbot] already exists.
[spawn_entity.py-4] [ERROR] [1711059164.544251656] [spawn_entity]: Spawn service failed. Exiting.
[ERROR] [spawn_entity.py-4]: process has died [pid 102276, exit code 1, cmd '/opt/ros/humble/lib/gazebo_ros/spawn_entity.py -topic robot_description -entity lidarbot --ros-args'].
[spawner-5] [INFO] [1711059166.424500117] [spawner_diff_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1711059166.437441751] [spawner_joint_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1711059168.436912695] [spawner_diff_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1711059168.449242074] [spawner_joint_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1711059170.450660089] [spawner_diff_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1711059170.461541510] [spawner_joint_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1711059172.468832480] [spawner_diff_controller]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1711059172.475391629] [spawner_joint_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-5] [ERROR] [1711059174.484462594] [spawner_diff_controller]: Controller manager not available
[spawner-6] [ERROR] [1711059174.492929827] [spawner_joint_broadcaster]: Controller manager not available
[ERROR] [spawner-6]: process has died [pid 102280, exit code 1, cmd '/home/noobinventor/ros2_ws/install/controller_manager/lib/controller_manager/spawner joint_broadcaster --ros-args'].
[ERROR] [spawner-5]: process has died [pid 102278, exit code 1, cmd '/home/noobinventor/ros2_ws/install/controller_manager/lib/controller_manager/spawner diff_controller --ros-args'].
```

saikishor, 2024-03-21:
Hello!

I've just tested [`example_9`](https://control.ros.org/humble/doc/ros2_control_demos/example_9/doc/userdoc.html) from [ros2_control_demos](https://github.com/ros-controls/ros2_control_demos/tree/humble) in a clean Humble docker and it works with 2.40.0. I recommend checking the same example in your setup

[image]

```
~/ros2_control_demos_ws# ros2 launch ros2_control_demo_example_9 rrbot_gazebo_classic.launch.py gui:=true
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-03-21-22-44-10-904243-india-11640
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [11642]
[INFO] [gzclient-2]: process started with pid [11644]
[INFO] [robot_state_publisher-3]: process started with pid [11646]
[INFO] [spawn_entity.py-4]: process started with pid [11648]
[INFO] [spawner-5]: process started with pid [11650]
[INFO] [spawner-6]: process started with pid [11652]
[INFO] [rviz2-7]: process started with pid [11654]
[robot_state_publisher-3] [INFO] [1711061051.377532820] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1711061051.377622801] [robot_state_publisher]: got segment link1
[robot_state_publisher-3] [INFO] [1711061051.377633820] [robot_state_publisher]: got segment link2
[robot_state_publisher-3] [INFO] [1711061051.377640611] [robot_state_publisher]: got segment tool_link
[robot_state_publisher-3] [INFO] [1711061051.377646868] [robot_state_publisher]: got segment world
[rviz2-7] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[rviz2-7] [INFO] [1711061051.498098803] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-7] [INFO] [1711061051.498322724] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[rviz2-7] [INFO] [1711061051.522762620] [rviz2]: Stereo is NOT SUPPORTED
[spawn_entity.py-4] [INFO] [1711061051.642388978] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1711061051.642727894] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-4] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-4]   warnings.warn(
[spawn_entity.py-4] [INFO] [1711061051.644648404] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-4] [INFO] [1711061051.655397798] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1711061051.655657484] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1711061052.409551510] [spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-4] [INFO] [1711061052.515255625] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [rrbot_system_position]
[gzserver-1] [INFO] [1711061052.532233831] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gzserver-1] [INFO] [1711061052.534606226] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /
[gzserver-1] [INFO] [1711061052.534806020] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gzserver-1] [INFO] [1711061052.534842337] [gazebo_ros2_control]: Loading parameter files /root/ros2_control_demos_ws/install/ros2_control_demo_example_9/share/ros2_control_demo_example_9/config/rrbot_controllers.yaml
[gzserver-1] [INFO] [1711061052.537328592] [gazebo_ros2_control]: connected to service!! robot_state_publisher
[gzserver-1] [INFO] [1711061052.537741196] [gazebo_ros2_control]: Received urdf from param server, parsing...
[gzserver-1] [INFO] [1711061052.548299320] [gazebo_ros2_control]: Loading joint: joint1
[gzserver-1] [INFO] [1711061052.548335585] [gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1711061052.548342304] [gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1711061052.548347864] [gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1711061052.548352211] [gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1711061052.548373891] [gazebo_ros2_control]: Loading joint: joint2
[gzserver-1] [INFO] [1711061052.548379629] [gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1711061052.548383822] [gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1711061052.548388101] [gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1711061052.548392251] [gazebo_ros2_control]: 		 position
[gzserver-1] [INFO] [1711061052.548424942] [resource_manager]: Initialize hardware 'RRBot' 
[gzserver-1] [INFO] [1711061052.548529965] [resource_manager]: Successful initialization of hardware 'RRBot'
[gzserver-1] [INFO] [1711061052.548593588] [resource_manager]: 'configure' hardware 'RRBot' 
[gzserver-1] [INFO] [1711061052.548599872] [resource_manager]: Successful 'configure' of hardware 'RRBot'
[gzserver-1] [INFO] [1711061052.548603678] [resource_manager]: 'activate' hardware 'RRBot' 
[gzserver-1] [INFO] [1711061052.548607554] [resource_manager]: Successful 'activate' of hardware 'RRBot'
[gzserver-1] [INFO] [1711061052.548656084] [gazebo_ros2_control]: Loading controller_manager
[gzserver-1] [WARN] [1711061052.556897260] [gazebo_ros2_control]:  Desired controller update period (0.1 s) is slower than the gazebo simulation period (0.001 s).
[gzserver-1] [INFO] [1711061052.557015736] [gazebo_ros2_control]: Loaded gazebo_ros2_control.
[gzserver-1] [INFO] [1711061052.606581327] [controller_manager]: Loading controller 'forward_position_controller'
[... cut ...]
[gzserver-1] [INFO] [1711061052.857287950] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[gzserver-1] [INFO] [1711061053.056083688] [forward_position_controller]: activate successful
[spawner-6] [INFO] [1711061053.157227902] [spawner_forward_position_controller]: Configured and activated forward_position_controller
[INFO] [spawner-6]: process has finished cleanly [pid 11652]
[spawner-5] [INFO] [1711061053.357608590] [spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster
[INFO] [spawner-5]: process has finished cleanly [pid 11650]
```

TheNoobInventor, 2024-03-21:
Okay thank you, I'll do that and let you know how that goes.

Ishansehgal, 2024-03-22:
I am getting this even when launching just the controller

the below is the output of the following launch file  

import launch
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, LogInfo
from launch_ros.actions import Node

def generate_launch_description():
    # Define launch arguments
    controller_name_arg = DeclareLaunchArgument('controller_name', default_value='joint_trajectory_controller')

    return LaunchDescription([
        controller_name_arg,
        LogInfo(msg='Loading controller...'),
        Node(
            package='controller_manager',
            executable='spawner',
            name='spawner_mani',
            namespace='',
            output='screen',
            arguments=['joint_trajectory_controller'],  # Specify the controller name here
        ),
        LogInfo(msg='Controller loaded successfully.'),
    ])

this is the output 

[INFO] [launch]: All log files can be found below /home/techturners/.ros/log/2024-03-22-11-50-20-989313-team1381-27606
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: Loading controller...
[INFO] [launch.user]: Controller loaded successfully.
[INFO] [spawner-1]: process started with pid [27613]
[spawner-1] [INFO] [1711088423.355662630] [spawner_mani]: Waiting for '/controller_manager' services to be available
[spawner-1] [INFO] [1711088425.372917947] [spawner_mani]: Waiting for '/controller_manager' services to be available
[spawner-1] [INFO] [1711088427.389829261] [spawner_mani]: Waiting for '/controller_manager' services to be available
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)

Even when I am running the ros2_control node in the background

techturners@team1381:~$ ros2 run controller_manager ros2_control_node
[WARN] [1711087969.617316492] [controller_manager]: 'update_rate' parameter not set, using default value.
[INFO] [1711087969.617422225] [controller_manager]: Subscribing to '~/robot_description' topic for robot description file.
[INFO] [1711087969.618065218] [controller_manager]: update rate is 100 Hz
[INFO] [1711087969.618302661] [controller_manager]: RT kernel is recommended for better performance

saikishor, 2024-03-22:
Ok I see the issue, you are not setting the `robot_description` parameter while launching the ros2_control_node or you should be publishing the robot_description to the topic namespaced by the node. This should solve the issue.

Did you try launching the above ros2_control_demos example 9 and check if it is working for you?

Thank you

Ishansehgal, 2024-03-22:
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [joint_state_publisher_gui-1]: process started with pid [96183]
[INFO] [robot_state_publisher-2]: process started with pid [96185]
[INFO] [rviz2-3]: process started with pid [96187]
[robot_state_publisher-2] [INFO] [1711093461.947908157] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1711093461.948112718] [robot_state_publisher]: got segment link1
[robot_state_publisher-2] [INFO] [1711093461.948124150] [robot_state_publisher]: got segment link2
[robot_state_publisher-2] [INFO] [1711093461.948132035] [robot_state_publisher]: got segment tool_link
[robot_state_publisher-2] [INFO] [1711093461.948139790] [robot_state_publisher]: got segment world
[rviz2-3] [INFO] [1711093462.703873073] [rviz2]: Stereo is NOT SUPPORTED
[joint_state_publisher_gui-1] [INFO] [1711093462.703964077] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
[rviz2-3] [INFO] [1711093462.704179489] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[joint_state_publisher_gui-1] [INFO] [1711093462.725028724] [joint_state_publisher]: Centering
[rviz2-3] [INFO] [1711093462.750118294] [rviz2]: Stereo is NOT SUPPORTED
[joint_state_publisher_gui-1] [INFO] [1711093465.600318176] [joint_state_publisher]: Centering
[joint_state_publisher_gui-1] [INFO] [1711093479.421412275] [joint_state_publisher]: Randomizing
[joint_state_publisher_gui-1] [INFO] [1711093588.995744435] [joint_state_publisher]: Centering

The demo is working fine, but I have a question: do I need to have a proper URDF to run the controller manager? I am still new to this and am unable to pinpoint the problem. My urdf file does not have the ros2 tags, but I am using my urdf and putting it into the Moveit setup assistance and adding the ros2_control configuration files there, and I do not know if I have to add the ros2_control tags in my urdf as well. Sorry if this is a silly question, but please clarify. and if i do have to how should i move forward

saikishor, 2024-03-22:
> Previously, my control manager was working fine and loading properly, but now that I have updated my packages,

@Ishansehgal If your don't have ros2_control tags, how come did you make it work in the first place?

Try to go through the [demos](https://control.ros.org/humble/doc/ros2_control_demos/doc/index.html) in the [ros2_control_demos](https://github.com/ros-controls/ros2_control_demos/tree/humble) package and try to understand how to setup your URDF to have it working. 

I'll be closing the issue as the issue is not within the packages. Good luck :)

Thank you,

closed 2024-03-22 by saikishor

Ishansehgal, 2024-03-22:
After publishing the robot_description (with ros2_control tags ) i ading the following in 
    spawn_controllers_manipulator = Node(
            package="controller_manager", 
            executable="spawner",
            name="spawner_mani",
            arguments=['joint_trajectory_controller'],
            output="screen")
    

    spawn_controllers_state = Node(
            package="controller_manager", 
            executable="spawner",
            arguments=['joint_state_broadcaster'],
            output="screen")
            
             the veiw_robot.launch.py  and again i got the same error 
             
             techturners@team1381:~$ ros2 launch ros2_control_demo_example_9 view_robot.launch.py
[INFO] [launch]: All log files can be found below /home/techturners/.ros/log/2024-03-22-15-06-26-343458-team1381-173727
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [joint_state_publisher_gui-1]: process started with pid [173736]
[INFO] [spawner-2]: process started with pid [173738]
[INFO] [spawner-3]: process started with pid [173740]
[INFO] [rviz2-4]: process started with pid [173742]
[joint_state_publisher_gui-1] [INFO] [1711100186.987290187] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
[joint_state_publisher_gui-1] [INFO] [1711100186.996412598] [joint_state_publisher]: Centering
[rviz2-4] [INFO] [1711100187.038336048] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [INFO] [1711100187.038500234] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-4] [INFO] [1711100187.059896087] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [ERROR] [1711100187.178490854] [rviz2]: The link is static or has unrealistic inertia, so the equivalent inertia box will not be shown.
[rviz2-4] 
[spawner-2] [INFO] [1711100188.748216280] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1711100188.845700944] [spawner_mani]: Waiting for '/controller_manager' node to exist
[joint_state_publisher_gui-1] [INFO] [1711100189.788433032] [joint_state_publisher]: Centering
[spawner-2] [INFO] [1711100190.762739104] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1711100190.857559147] [spawner_mani]: Waiting for '/controller_manager' node to exist
[spawner-2] [INFO] [1711100192.774381456] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1711100192.869700382] [spawner_mani]: Waiting for '/controller_manager' node to exist
[spawner-2] [INFO] [1711100194.787677957] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1711100194.881494494] [spawner_mani]: Waiting for '/controller_manager' node to exist
[spawner-2] [ERROR] [1711100196.799561279] [spawner_joint_state_broadcaster]: Controller manager not available
[spawner-3] [ERROR] [1711100196.893426685] [spawner_mani]: Controller manager not available
[ERROR] [spawner-2]: process has died [pid 173738, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_state_broadcaster --ros-args'].
[ERROR] [spawner-3]: process has died [pid 173740, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_trajectory_controller --ros-args -r __node:=spawner_mani'].
^C[WARNING] [launch]: user interrupted with 
             
sorry but i am stuck at this for too long and not able to figure out

christophfroehlich, 2024-03-22:
Please take your time to properly format your comments. Have you launched a ros2_control_node at all? There is no evidence in your log file. As Sai has written above, have a look into our demos how to setup this framework.

Ishansehgal, 2024-03-22:
ok thaks i will get back to you

Ishansehgal, 2024-03-23:
this is the log of the code which was running before the updates and it shows the below error now 

~/ros2_ws$ ros2 launch ur5_moveit spawn_ur5_launch_moveit.launch.py
[INFO] [launch]: All log files can be found below /home/kanish/.ros/log/2024-03-22-15-56-30-186262-team1381-364825
[INFO] [launch]: Default logging verbosity is set to INFO
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
[INFO] [robot_state_publisher-1]: process started with pid [364843]
[robot_state_publisher-1] [INFO] [1711103190.964018384] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1711103190.964375585] [robot_state_publisher]: got segment camera_bottom_screw_frame
[robot_state_publisher-1] [INFO] [1711103190.964399422] [robot_state_publisher]: got segment camera_color_frame
[robot_state_publisher-1] [INFO] [1711103190.964409492] [robot_state_publisher]: got segment camera_color_optical_frame
[robot_state_publisher-1] [INFO] [1711103190.964417180] [robot_state_publisher]: got segment camera_depth_frame
[robot_state_publisher-1] [INFO] [1711103190.964424675] [robot_state_publisher]: got segment camera_depth_optical_frame
[robot_state_publisher-1] [INFO] [1711103190.964432162] [robot_state_publisher]: got segment camera_left_ir_frame
[robot_state_publisher-1] [INFO] [1711103190.964439982] [robot_state_publisher]: got segment camera_left_ir_optical_frame
[robot_state_publisher-1] [INFO] [1711103190.964447875] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-1] [INFO] [1711103190.964455137] [robot_state_publisher]: got segment camera_link_depth
[robot_state_publisher-1] [INFO] [1711103190.964461891] [robot_state_publisher]: got segment camera_right_ir_frame
[robot_state_publisher-1] [INFO] [1711103190.964469563] [robot_state_publisher]: got segment camera_right_ir_optical_frame
[robot_state_publisher-1] [INFO] [1711103190.964477279] [robot_state_publisher]: got segment ee_link
[robot_state_publisher-1] [INFO] [1711103190.964484982] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-1] [INFO] [1711103190.964492744] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-1] [INFO] [1711103190.964500661] [robot_state_publisher]: got segment tool0
[robot_state_publisher-1] [INFO] [1711103190.964508453] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-1] [INFO] [1711103190.964516420] [robot_state_publisher]: got segment world
[robot_state_publisher-1] [INFO] [1711103190.964524344] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-1] [INFO] [1711103190.964532134] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-1] [INFO] [1711103190.964539778] [robot_state_publisher]: got segment wrist_3_link
[INFO] [spawner-2]: process started with pid [364852]
[INFO] [spawner-3]: process started with pid [364854]
[INFO] [rviz2-4]: process started with pid [364856]
[INFO] [ros2_control_node-5]: process started with pid [364858]
[INFO] [move_group-6]: process started with pid [364860]
[INFO] [servo_node_main-7]: process started with pid [364862]
[rviz2-4] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[ros2_control_node-5] [WARN] [1711103192.289095063] [control_node_ros2]: 'update_rate' parameter not set, using default value.
[ros2_control_node-5] [WARN] [1711103192.289242661] [control_node_ros2]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-5] [INFO] [1711103192.289650112] [resource_manager]: Loading hardware 'FakeSystem' 
[ros2_control_node-5] [INFO] [1711103192.290247934] [resource_manager]: Initialize hardware 'FakeSystem' 
[ros2_control_node-5] [WARN] [1711103192.290310597] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-5] <state_interface name="velocity"> 
[ros2_control_node-5]   <param name="initial_value">0.0</param> 
[ros2_control_node-5] </state_interface>
[ros2_control_node-5] [INFO] [1711103192.290329412] [resource_manager]: Successful initialization of hardware 'FakeSystem'
[ros2_control_node-5] [INFO] [1711103192.290387996] [resource_manager]: 'configure' hardware 'FakeSystem' 
[ros2_control_node-5] [INFO] [1711103192.290400534] [resource_manager]: Successful 'configure' of hardware 'FakeSystem'
[ros2_control_node-5] [INFO] [1711103192.290409435] [resource_manager]: 'activate' hardware 'FakeSystem' 
[ros2_control_node-5] [INFO] [1711103192.290416442] [resource_manager]: Successful 'activate' of hardware 'FakeSystem'
[ros2_control_node-5] [INFO] [1711103192.321275169] [control_node_ros2]: update rate is 100 Hz
[ros2_control_node-5] [INFO] [1711103192.343816539] [control_node_ros2]: RT kernel is recommended for better performance
[move_group-6] [WARN] [1711103192.353732807] [move_group.move_group]: Using default pipeline 'ompl'
[move_group-6] [INFO] [1711103192.363081225] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-6] [INFO] [1711103192.363129308] [moveit_robot_model.robot_model]: Loading robot model 'ur5'...
[move_group-6] [WARN] [1711103192.373833037] [moveit_robot_model.robot_model]: Could not identify parent group for end-effector 'gripper'
[... cut ...] open.
[rviz2-4]          at line 253 in /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp
[spawner-3] [INFO] [1711103194.938250953] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-2] [INFO] [1711103195.479310147] [spawner_mani]: Waiting for '/controller_manager' services to be available
[spawner-3] [INFO] [1711103196.947556779] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available

christophfroehlich, 2024-03-23:
Please take your time to properly format your comments. 
It seems that you have remapped the name of the controller manger to `control_node_ros2`. You can check with `ros2 service list`, the services will be in the `control_node_ros2` namespace. Remove this, or add the controller manager name to the [spawner](https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html#spawner).

TheNoobInventor, 2024-03-23:
@saikishor I apologize for the late response, I've been occupied lately. I wasn't able to build the `gazebo_ros2_control` demos however, I looked at how everything was setup in the [diff drive launch file](https://github.com/ros-controls/gazebo_ros2_control/blob/master/gazebo_ros2_control_demos/launch/diff_drive.launch.py) and I got it to work:

```
[spawn_entity.py-4] [INFO] [1711192330.657839650] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [lidarbot]
[gzserver-2] [INFO] [1711192330.674094689] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gzserver-2] [INFO] [1711192330.678667393] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /
[gzserver-2] [INFO] [1711192330.678730565] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gzserver-2] [INFO] [1711192330.678782224] [gazebo_ros2_control]: Loading parameter files /home/noobinventor/lidarbot_ws/install/lidarbot_bringup/share/lidarbot_bringup/config/controllers.yaml
[gzserver-2] [INFO] [1711192330.678793879] [gazebo_ros2_control]: Loading parameter files /home/noobinventor/lidarbot_ws/install/lidarbot_bringup/share/lidarbot_bringup/config/gazebo_ros2_ctl_use_sim.yaml
[gzserver-2] [INFO] [1711192330.682024423] [gazebo_ros2_control]: connected to service!! robot_state_publisher
[gzserver-2] [INFO] [1711192330.682675940] [gazebo_ros2_control]: Received urdf from param server, parsing...
[gzserver-2] [INFO] [1711192330.693052861] [gazebo_ros2_control]: Loading joint: left_wheel_joint
[gzserver-2] [INFO] [1711192330.693108451] [gazebo_ros2_control]:       State:
[gzserver-2] [INFO] [1711192330.693115632] [gazebo_ros2_control]:                velocity
[gzserver-2] [INFO] [1711192330.693121569] [gazebo_ros2_control]:                position
[gzserver-2] [INFO] [1711192330.693125749] [gazebo_ros2_control]:       Command:
[gzserver-2] [INFO] [1711192330.693148025] [gazebo_ros2_control]:                velocity
[gzserver-2] [INFO] [1711192330.693155840] [gazebo_ros2_control]: Loading joint: right_wheel_joint
[gzserver-2] [INFO] [1711192330.693159775] [gazebo_ros2_control]:       State:
[gzserver-2] [INFO] [1711192330.693163124] [gazebo_ros2_control]:                velocity
[gzserver-2] [INFO] [1711192330.693166469] [gazebo_ros2_control]:                position
[gzserver-2] [INFO] [1711192330.693169671] [gazebo_ros2_control]:       Command:
[gzserver-2] [INFO] [1711192330.693174838] [gazebo_ros2_control]:                velocity
[gzserver-2] [INFO] [1711192330.693212649] [resource_manager]: Initialize hardware 'GazeboSystem'
[gzserver-2] [INFO] [1711192330.693393808] [resource_manager]: Successful initialization of hardware 'GazeboSystem'
[gzserver-2] [INFO] [1711192330.693467671] [resource_manager]: 'configure' hardware 'GazeboSystem'
[gzserver-2] [INFO] [1711192330.693475151] [resource_manager]: Successful 'configure' of hardware 'GazeboSystem'
[gzserver-2] [INFO] [1711192330.693480083] [resource_manager]: 'activate' hardware 'GazeboSystem'
[gzserver-2] [INFO] [1711192330.693484265] [resource_manager]: Successful 'activate' of hardware 'GazeboSystem'
[gzserver-2] [INFO] [1711192330.693550891] [gazebo_ros2_control]: Loading controller_manager
[gzserver-2] [WARN] [1711192330.711146180] [gazebo_ros2_control]:  Desired controller update period (0.0333333 s) is slower than the gazebo simulation period (0.001 s).
[gzserver-2] [INFO] [1711192330.711372149] [gazebo_ros2_control]: Loaded gazebo_ros2_control.
```
I did not know that things had shifted from calling the `spawner` executable to using `load_controller` executable which might be perfect for my purposes because I previously had to introduce a time delay to load everything in the right order.

Thank you @saikishor and @christophfroehlich!

christophfroehlich, 2024-03-23:
> I did not know that things had shifted from calling the `spawner` executable to using `load_controller` executable which might be perfect for my purposes because I previously had to introduce a time delay to load everything in the right order.
> 

That's not correct, both should be supported. See the [example_9](https://github.com/ros-controls/ros2_control_demos/blob/master/example_9/bringup/launch/rrbot_gazebo_classic.launch.py#L77-L87), where we use the spawner executable.

TheNoobInventor, 2024-03-23:
> That's not correct, both should be supported. See the [example_9](https://github.com/ros-controls/ros2_control_demos/blob/master/example_9/bringup/launch/rrbot_gazebo_classic.launch.py#L77-L87), [...]

Oh okay, I did not know that, I will check it out. Thanks again.

Ishansehgal, 2024-03-27:
Thank you, @christophfroehlich , for pointing out the problem. I was able to resolve it by carefully examining the code. I have a question regarding the behavior change after the update.

Before the update, the code block with the name='control_node_ros2' line worked perfectly fine. However, after the update, commenting out this line seemed to resolve the issue.

I'm curious about why specifying the name (name=) in this context affects the node's behavior. Could you provide insights into why this change occurred after the update? It's puzzling because the code structure remained the same, and I'm confident that I'm using the same code.

Thank you for your assistance!
[image]

christophfroehlich, 2024-04-04:
Do you know which version you had before? Since PR #1052 introduces with 2.35.0, the CM subscribes to the `~/robot_description` topic. Setting a name of the node changes the namespace of the topic, this might be the cause here.

Ishansehgal, 2024-04-06:
Thanks Alot for the clarifictaion