ros-controls/ros2_control #1466: Joint Limiters: Saturation (current PR971 - will be renamed) and Soft Limiter

bmagyar, 2024-04-01:
ros-controls/ros2_control#971

1. PR: `Simple Joint Limiter` --> `Saturation Joint Limiter`  --> Done in ros-controls/ros2_control#971
    - check ros_control implementation for full functionality
    - add also saturation of the `effort` interface --> 
2. PR: Parse URDF data and use `max_acceleration` and `max_deceleration` from `<ros2_control>` Tag
4. PR: Open PR Proposal for extending URDF with these values
5. PR: Add `Soft Joint Limiter` (new PR)

bmagyar, 2024-04-01:
Related to #744

saikishor, 2024-04-02:
PR: https://github.com/ros-controls/ros2_control/pull/1472 handles the point 2 of this issue, and there will be another upcoming PR adding the SoftLimits to the `HardwareInfo` structure

saikishor, 2024-06-17:
We now have this https://github.com/ros-controls/ros2_control/pull/1526 that includes the changes of #971 and also modified in such a way that it fits in with the integration into the ResourceManager to be able to enforce limits directly over there.

The integration into the ResourceManager would look something like this : https://github.com/saikishor/ros2_control/pull/4

closed 2025-04-25 by christophfroehlich