ros-controls/ros2_control #1480: not able to activate 2 controllers

FrGe2016, 2024-04-14:
Not realy a bug probably just missunderstanding.

I have a homemade robotic arm. IT is controlled by 2 ESP32 one for the arm the other for the gripper. Based on denis.stogl i have made a hardware interface named FBotHardwareInterface. I have tried to have only one plugin to maintain, so based on the number of joints in the controller,  i have specific code to read states and send commands with the good ESP32.

When i launch my file , every thing looks good:
      Loaded joint_state_broadcaster
      Configured and activated joint_state_broadcaster
      Loaded arm_controller
      Loaded gripper_controller

Then :
ros2_control_node-1] [INFO] [1713113108.601460156] [controller_manager]: Configuring controller 'gripper_controller'
[ros2_control_node-1] [INFO] [1713113108.601755394] [gripper_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[ros2_control_node-1] [INFO] [1713113108.601820418] [gripper_controller]: Command interfaces are [position] and state interfaces are [position velocity].
[ros2_control_node-1] [INFO] [1713113108.601851167] [gripper_controller]: Using 'splines' interpolation method.
[ros2_control_node-1] [INFO] [1713113108.602381975] [gripper_controller]: Controller state will be published at 50.00 Hz.
[ros2_control_node-1] [INFO] [1713113108.605310772] [gripper_controller]: Goals with partial set of joints are allowed
[ros2_control_node-1] [INFO] [1713113108.605346811] [gripper_controller]: Action status changes will be monitored at 20.00 Hz.
[ros2_control_node-1] [ERROR] [1713113108.651660386] [resource_manager]: Not acceptable command interfaces combination: 
[ros2_control_node-1] Start interfaces: 
[ros2_control_node-1] [
[ros2_control_node-1]   elbow/position
[ros2_control_node-1]   wrist/position
[ros2_control_node-1]   rotator/position
[ros2_control_node-1]   lateral/position
[ros2_control_node-1] ]
[ros2_control_node-1] Stop interfaces: 
[ros2_control_node-1] [
[ros2_control_node-1] ]
[ros2_control_node-1] Not existing: 
[ros2_control_node-1] [
[ros2_control_node-1]  elbow/position
[ros2_control_node-1]  wrist/position
[ros2_control_node-1]  rotator/position
[ros2_control_node-1]  lateral/position
[ros2_control_node-1] ]
[ros2_control_node-1] 
[ros2_control_node-1] [ERROR] [1713113108.651702656] [controller_manager]: Could not switch controllers since prepare command mode switch was rejected.
[spawner-5] [ERROR] [1713113108.653948356] [spawner_arm_controller]: Failed to activate controller
[spawner-4] [INFO] [1713113108.752554543] [spawner_gripper_controller]: Configured and activated gripper_controller

Here are the associated files:

g_controllers.yaml

# This config file is used by ros2_control
controller_manager:
  ros__parameters:                            # 'est ce fichier qui est utilisé dans d2.real
    update_rate: 20  # Hz

    arm_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    gripper_controller:
      type: joint_trajectory_controller/JointTrajectoryController  
     
    joint_state_broadcaster:                                   
      type: joint_state_broadcaster/JointStateBroadcaster      
     
arm_controller:
  ros__parameters:
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
    allow_partial_joints_goal: true
    joints:
      - elbow
      - wrist
      - rotator
      - lateral

gripper_controller:
  ros__parameters:
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
    allow_partial_joints_goal: true
    joints:
      - gripper
      - gripper_top

g.ros2_control_arm.xacro

<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

  <xacro:macro name="g_ros2_control" params="
               name
               use_mock_hardware:=^|false
               mock_sensor_commands:=^|false
               sim_gazebo_classic:=^|false
               sim_gazebo:=^|false"
               >
    <ros2_control name="${name}" type="system">
      <hardware>
        <xacro:if value="${use_mock_hardware}">
          <plugin>mock_components/GenericSystem</plugin>
          <param name="mock_sensor_commands">${mock_sensor_commands}</param>
        </xacro:if>
        <xacro:if value="${sim_gazebo_classic}">
          <plugin>gazebo_ros2_control/GazeboSystem</plugin>
        </xacro:if>
        <xacro:if value="${sim_gazebo}">
          <plugin>ign_ros2_control/IgnitionSystem</plugin>
        </xacro:if>
        <xacro:unless value="${use_mock_hardware or sim_gazebo_classic or sim_gazebo}">
          <plugin>f_hardware_interface/FBotHardwareInterface</plugin>
                  
        </xacro:unless>
      </hardware>
      <joint name="elbow">
         <command_interface name="position"/>
         <command_interface name="velocity"/>
         <state_interface name="position">
             <param name="initial_value">0.0</param>
          </state_interface>
          <state_interface name="velocity">
             <param name="initial_value">0.0</param>
          </state_interface>    
      </joint>
      <joint name="wrist">
         <command_interface name="position"/>
         <command_interface name="velocity"/>
         <state_interface name="position">
             <param name="initial_value">0.0</param>
          </state_interface>
          <state_interface name="velocity">
             <param name="initial_value">0.0</param>
          </state_interface> 
      </joint>
      <joint name="rotator">
         <command_interface name="position"/>
         <command_interface name="velocity"/>
         <state_interface name="position">
             <param name="initial_value">0.0</param>
          </state_interface>
          <state_interface name="velocity">
             <param name="initial_value">0.0</param>
          </state_interface> 
      </joint>
      <joint name="lateral">
         <command_interface name="position"/>
         <state_interface name="position">
             <param name="initial_value">0.0</param>
          </state_interface>
          <state_interface name="velocity">
             <param name="initial_value">0.0</param>
          </state_interface> 
      </joint>

    </ros2_control>
    

         
    
    g.ros2_control_arm.xacro
    
  </xacro:macro>
</robot>

<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

  <xacro:macro name="g_ros2_control" params="
               name
               use_mock_hardware:=^|false
               mock_sensor_commands:=^|false
               sim_gazebo_classic:=^|false
               sim_gazebo:=^|false"
               >
    <ros2_control name="${name}" type="system">
      <hardware>
        <xacro:if value="${use_mock_hardware}">
          <plugin>mock_components/GenericSystem</plugin>
          <param name="mock_sensor_commands">${mock_sensor_commands}</param>
        </xacro:if>
        <xacro:if value="${sim_gazebo_classic}">
          <plugin>gazebo_ros2_control/GazeboSystem</plugin>
        </xacro:if>
        <xacro:if value="${sim_gazebo}">
          <plugin>ign_ros2_control/IgnitionSystem</plugin>
        </xacro:if>
        <xacro:unless value="${use_mock_hardware or sim_gazebo_classic or sim_gazebo}">
          <plugin>f_hardware_interface/FBotHardwareInterface</plugin>
                  
        </xacro:unless>
      </hardware>
      <joint name="gripper">
         <command_interface name="position"/>
         <command_interface name="velocity"/>
         <state_interface name="position">
             <param name="initial_value">0.0</param>
          </state_interface>
          <state_interface name="velocity">
             <param name="initial_value">0.0</param>
          </state_interface> 
      </joint>
      <joint name="gripper_top">
         <command_interface name="position"/>
         <command_interface name="velocity"/>
         <state_interface name="position">
             <param name="initial_value">0.0</param>
          </state_interface>
          <state_interface name="velocity">
             <param name="initial_value">0.0</param>
          </state_interface> 
      </joint>

    </ros2_control>
    

         
    
    g.ros2_control_gripper.xacro
    
  </xacro:macro>
</robot>

christophfroehlich, 2024-04-15:
First of all, please take your time to properly format your questions (I did that now for you).

The error means that your hardware component did not load the requested interfaces. Remove the controller spawners from your launch file, launch your system and have a look with `ros2 control list_hardware_components` and `ros2 control list_hardware_interfaces`

From your code snippets it seems that you override your macro `g_ros2_control` for the gripper if you load that in the same URDF. Try to run `xacro` manually and introspect the resulting URDF in an editor, is there `<plugin>f_grip_hardware_interface/FGripHardwareInterface</plugin>` and `<plugin>f_hardware_interface/FBotHardwareInterface</plugin>` for both components?

FrGe2016, 2024-04-16:
Thank for the urdf clue. Since both macro had the same name, the macro was called  only for one of the controller only.

FrGe2016, 2024-04-16:
Closed issue

closed 2024-04-16 by FrGe2016