ros-controls/ros2_control #1482: `KeyboardInterrupt` exception if shutdown before controller manager is found

Timple, 2024-04-16:
**Describe the bug**
`KeyboardInterrupt` exception if shutdown before controller manager is found

**To Reproduce**
Steps to reproduce the behavior:
1. `ros2 run controller_manager spawner test`
2. Press ctrl+C
3. See error

**Expected behavior**
Clean shutdown

**Screenshots**
```
[spawner-64] Traceback (most recent call last):
[spawner-64]   File "/opt/ros/iron/lib/controller_manager/spawner", line 33, in <module>
[spawner-64]     sys.exit(load_entry_point('controller-manager==3.24.0', 'console_scripts', 'spawner')())
[spawner-64]   File "/opt/ros/iron/lib/python3.10/site-packages/controller_manager/spawner.py", line 214, in main
[spawner-64]     if not wait_for_controller_manager(
[spawner-64]   File "/opt/ros/iron/lib/python3.10/site-packages/controller_manager/spawner.py", line 105, in wait_for_controller_manager
[spawner-64]     node_and_namespace = wait_for_value_or(
[spawner-64]   File "/opt/ros/iron/lib/python3.10/site-packages/controller_manager/spawner.py", line 65, in wait_for_value_or
[spawner-64]     time.sleep(0.2)
[spawner-64] KeyboardInterrupt
```

**Environment (please complete the following information):**
 - OS: `linux`
 - Version: `iron` but seems identical code on rolling

christophfroehlich, 2024-04-16:
I can confirm this issue. Would you mind taking a look into the script and fix it?

destogl, 2024-08-15:
resolved in #1562

closed 2024-08-15 by destogl