ros-controls/ros2_control #1486: Passing data to multiple instances of HardwareInterface from control node

tenfoldpaper, 2024-04-18:
Hi, I've been working on a mujoco integration of the Panda robot arm into ros2-control.
So far, I've had some success simulating multiple robots, but only by creating the simulation loop inside the HardwareInterface class itself. While I can control and do things with it, when it comes to combining multiple hardware interfaces together (e.g. the robot arm mounted on a mobile base with its own HardwareInterface class), this structure would require me to write a monolothic HardwareInterface class that combines both of these together.

I've taken a look at the linked mujoco integration repo, and it appears that they are also doing something similar.

From what I can tell, in ros-control, we had a more fine-grained control over how to initialize the Interface classes, as it had a constructor and we could initialize the HardwareInterface-equivalent with an argument to simulate multiple components all sharing the same data source.

Is this a feature that is also available in ros2-control? If so, how do you do this? Apologies in advance for asking something that may be basic, but I could not find any relevant code bases that address this issue.

christophfroehlich, 2024-04-24:
Do you need to share data at run-time, or only at loading/initialization of the different components?

tenfoldpaper, 2024-05-09:
> Do you need to share data at run-time, or only at loading/initialization of the different components?

Sorry for the late reply!
I need to share data at run-time. Generally, this should be possible by passing a reference to the mjData* and mjModel* pointers that get instantiated when loading up the scene using Mujoco. So far I've managed that successfully with member variable classes.

Since data is only accessed in the controller's read and write loops (and read and written in those parts), there shouldn't be major issues, as from what I can tell, the respective HardwareInterfaces' read and write functions are executed serially.

There is a mujoco-ros-control repo for ROS1 I found before that did the whole manual initialization thing, which made this possible. So far, I haven't been able to find a solution to this yet.

tenfoldpaper, 2024-05-09:
> @tenfoldpaper maybe implementation of `gz_ros2_control` is interesting in this regard for you? Can you check it?

I'll take a look at it in a couple days and update my comment on this, but thank you for the suggestion!

tenfoldpaper, 2024-05-15:
So I went through `gz_ros2_control`, and it appears that the way it is handling `ros2-control` is not really transferable to MuJoCo. 
It seems to work based on Gazebo's Plugin interface, which seems to create instances of ros2-control objects inside the simulation directly. 

Since MuJoCo needs to be initialized and executed directly in `ros2-control` in the main control node, and there is no easy way to create a similar plugin-like interface for it, unfortunately it does not solve the problem. 

@christophfroehlich , do you know any way to implement what I asked in the opening post?

christophfroehlich, 2024-05-15:
No I don't know a way to pass data between different hardware components. But maybe I don't understand your problem correctly: You will have to use different hardware components anyways, one (set) for your robot, and another one (set) for your simulation. For example, two hardware components for your real hardware (arm + base), but a single monolithic one for mujoco?

saikishor, 2024-05-15:
@tenfoldpaper  Maybe what you are looking for is a [`System`](https://github.com/ros-controls/ros2_control/blob/master/hardware_interface/include/hardware_interface/system_interface.hpp) [HardwareComponents](https://control.ros.org/rolling/doc/getting_started/getting_started.html#hardware-components), where multiple actuators can coexist and share information.

christophfroehlich, 2024-05-23:
> @tenfoldpaper Maybe what you are looking for is a [`System`](https://github.com/ros-controls/ros2_control/blob/master/hardware_interface/include/hardware_interface/system_interface.hpp) [...]

as far as I understood he wants to share data between components which are not known altogether at compile time, but loaded dynamically.

aialmohamed, 2024-10-15:
Hi, I am also trying to achieve something similar and have a question.
So I want to build a digital twin of a robot (a digital model with Gazebo Ignition and Kuka Robot).
As far as I know it is possible to use multiple controllers, but hardware plugins (under the ros2_control tag) are not possible (maybe I misunderstood).
So the question is, can we run multiple (ros2_control) tags in one URDF? and if so, can we share the joints between them?

christophfroehlich, 2024-10-15:
Do you mean that you want to run the digital twin simultaneously with your real robot, but want only one controller operating on both hardware components? That is not possible (and I don't understand the the use case of the digital twin tbh).

christophfroehlich, 2024-10-16:
I still don't get the idea, what should this "Bi-direction bridge" do? If both controller_managers are independent, why you don't just use xacro macros to enable one of the two hardware components, like we do in [this demo](https://github.com/ros-controls/ros2_control_demos/blob/master/example_9/description/ros2_control/rrbot.ros2_control.xacro)?

aialmohamed, 2024-10-16:
@christophfroehlich 
i'm simply trying to run the simulation and the real robot simultaneously and let the simulation lets say to "mirror" the real robot in a physical world .
I will take a look on this Demo .
Thanks alot

saikishor, 2024-10-17:
> @christophfroehlich 
> i'm simply trying to run the simulation and the real robot simultaneously and let the simulation lets say to "mirror" the real robot in a physical world .
> I will take a look on this Demo .
> Thanks alot

Hello @aialmohamed 

Why do you want the simulation mirror the real robot? Visualization purposes?. If so, why don't you use RViZ2?

closed 2025-05-16 by github-actions[bot]