ros-controls/ros2_control #1506: spawner.py parse parameters with namespace wrong

delihus, 2024-04-25:
**Describe the bug**
Using spawner.py with `--params-file` and `--namespace` doesn't work properly. 

**To Reproduce**
Steps to reproduce the behavior:

1. Launch controller manager with namespace `panther`

2. Take this file:
```yaml 
/**:
  controller_manager:
    ros__parameters:
      update_rate: 500  # Hz

      joint_state_broadcaster:
        type: joint_state_broadcaster/JointStateBroadcaster

      io_and_status_controller:
        type: ur_controllers/GPIOController

      speed_scaling_state_broadcaster:
        type: ur_controllers/SpeedScalingStateBroadcaster

      force_torque_sensor_broadcaster:
        type: ur_controllers/ForceTorqueStateBroadcaster

      joint_trajectory_controller:
        type: joint_trajectory_controller/JointTrajectoryController

      scaled_joint_trajectory_controller:
        type: ur_controllers/ScaledJointTrajectoryController

      forward_velocity_controller:
        type: velocity_controllers/JointGroupVelocityController

      forward_position_controller:
        type: position_controllers/JointGroupPositionController

  speed_scaling_state_broadcaster:
    ros__parameters:
      state_publish_rate: 100.0

  force_torque_sensor_broadcaster:
    ros__parameters:
      sensor_name: tcp_fts_sensor
      state_interface_names:
        - force.x
        - force.y
        - force.z
        - torque.x
        - torque.y
        - torque.z
      frame_id: tool0
      topic_name: ft_data

  joint_trajectory_controller:
    ros__parameters:
      joints:
        - shoulder_pan_joint
        - shoulder_lift_joint
        - elbow_joint
        - wrist_1_joint
        - wrist_2_joint
        - wrist_3_joint
      command_interfaces:
        - position
      state_interfaces:
        - position
        - velocity
      state_publish_rate: 100.0
      action_monitor_rate: 20.0
      allow_partial_joints_goal: false
      constraints:
        stopped_velocity_tolerance: 0.2
        goal_time: 0.0
        shoulder_pan_joint: { trajectory: 0.2, goal: 0.1 }
        shoulder_lift_joint: { trajectory: 0.2, goal: 0.1 }
        elbow_joint: { trajectory: 0.2, goal: 0.1 }
        wrist_1_joint: { trajectory: 0.2, goal: 0.1 }
        wrist_2_joint: { trajectory: 0.2, goal: 0.1 }
        wrist_3_joint: { trajectory: 0.2, goal: 0.1 }

  scaled_joint_trajectory_controller:
    ros__parameters:
      joints:
        - shoulder_pan_joint
        - shoulder_lift_joint
        - elbow_joint
        - wrist_1_joint
        - wrist_2_joint
        - wrist_3_joint
      command_interfaces:
        - position
      state_interfaces:
        - position
        - velocity
      state_publish_rate: 100.0
      action_monitor_rate: 20.0
      allow_partial_joints_goal: false
      constraints:
        stopped_velocity_tolerance: 0.2
        goal_time: 0.0
        shoulder_pan_joint: { trajectory: 0.2, goal: 0.1 }
        shoulder_lift_joint: { trajectory: 0.2, goal: 0.1 }
        elbow_joint: { trajectory: 0.2, goal: 0.1 }
        wrist_1_joint: { trajectory: 0.2, goal: 0.1 }
        wrist_2_joint: { trajectory: 0.2, goal: 0.1 }
        wrist_3_joint: { trajectory: 0.2, goal: 0.1 }

  forward_velocity_controller:
    ros__parameters:
      joints:
        - shoulder_pan_joint
        - shoulder_lift_joint
        - elbow_joint
        - wrist_1_joint
        - wrist_2_joint
        - wrist_3_joint
      interface_name: velocity

  forward_position_controller:
    ros__parameters:
      joints:
        - shoulder_pan_joint
        - shoulder_lift_joint
        - elbow_joint
        - wrist_1_joint
        - wrist_2_joint
        - wrist_3_joint

```
2. Save it to `/tmp/jtc.yaml`

3. Spawn JTC:
```bash
python3 spawner.py  joint_trajectory_controller -c controller_manager --controller-manager-timeout 10 --namespace panther --param-file /tmp/jtc.yaml --ros-args -r __ns:=/panther -p use_sim_time:=True
```
4. See error:
```bash
/home/deli/Documents/panther_dev/spawner.py:401: DeprecationWarning: 'spawner.py' is deprecated, please use 'spawner' (without .py extension)
[INFO] [1714054504.062935319] [panther.spawner_joint_trajectory_controller]: Set controller params file to "/tmp/jtc.yaml" for panther/joint_trajectory_controller
[FATAL] [1714054504.064816747] [panther.spawner_joint_trajectory_controller]: Failed loading controller panther/joint_trajectory_controller
```

**Expected behavior**
JTC should be configured correctly ans spawner.py should correctly parse parameters.

**Screenshots**
The red is wrong parameter.  The green is good after this fix:
https://github.com/ros-controls/ros2_control/blob/humble/controller_manager/controller_manager/spawner.py#L272
FROM:
```python
parameter.name = prefixed_controller_name + ".params_file"
```
TO:
```python
parameter.name = controller_name + ".params_file"
```

[image]

ERROR
[image]
**Environment (please complete the following information):**
 - OS: ubuntu 22.04
 - Version humble

christophfroehlich, 2024-05-17:
It seems that you have already found a fix, would you mind creating a PR for this?

delihus, 2024-05-23:
@christophfroehlich Created PR https://github.com/ros-controls/ros2_control/pull/1547

PR #1640 (2024-07-29): Fix the namespaced controller_manager spawner + added tests
This PR aims to fix the spawner to work with namespace controller manager and added tests for this particular case. It is started from #1547, however after looking into the spawner, it seems like the `--namespace` parameter is no longer required as you could pass it in the arg `--controller-manager` as how it is done for the unspawner, if this is fine, then we can get rid of the `--namespace` arg 

closes: #1547
fixes: #1506
closed 2024-07-29 by bmagyar