ros-controls/ros2_control #1527: [Question / Feature Request] - Is there a way to list _all_ controllers (not just loaded controllers)

firesurfer, 2024-05-08:
Hi, 

I am looking for a way to list _all_ controllers that are known to ros2control, not just loaded ones as 
`ros2 control list_controllers` does. 

By all controllers I mean all the controllers that are defined in the `controllers.yaml` file. 

Thanks a lot in advance!

fmauch, 2024-05-08:
I was asking myself the same thing recently, also given that the rqt plugin shows all of them.

saikishor, 2024-05-08:
I think the rqt_plugin uses the info from the `list_controllers` topic
https://github.com/ros-controls/ros2_control/blob/607755e89636efdd13f6d70ebc07376f9020a820/rqt_controller_manager/rqt_controller_manager/controller_manager.py#L152-L162

I'm not sure if there is a way to read it and display it, I'm not sure if this feature is necessary, as in the future we might want to move the definition of the controller types from the original yaml that is parsed to the controller_manager node (https://github.com/ros-controls/ros2_control_demos/pull/502), in that case, it would be hard to read the all available controller. However, what can be seen is the available controller types.

firesurfer, 2024-05-08:
As far as I can tell the `list_controllers` topic only lists loaded controllers but not defined controllers. 

> I'm not sure if there is a way to read it and display it, I'm not sure if this feature is necessary, as in the future we might want to move the definition of the controller types from the original [...]

Well at some point the controller manager has to know about available controllers, otherwise it would not be possible to load them. So it should also be possible to expose them to the user. 

The reason I want to access them is that I define a list of controllers in our application framework which then can be used by users for certain operations. At startup I would like to check if that list matches with the controllers known to the controllers manager.

saikishor, 2024-05-08:
> Well at some point the controller manager has to know about available controllers, otherwise it would not be possible to load them. So it should also be possible to expose them to the user.

Well not necessarily, it is indeed possible to load them dynamically(https://github.com/ros-controls/ros2_control/pull/1502 and https://github.com/ros-controls/ros2_control/pull/1293). I have added this feature very recently. The controller manager doesn't have to know the name of the controller, the user should be able to load any type of controller with whatever name he wants. 

If you want to verify the list, why don't you spawn all the controllers with `--load-only` option and then just validate the list, and you could be able to activate and deactivate from there on.

firesurfer, 2024-05-08:
Okey it somehow makes sense if it is possible to load controller at runtime. Even though from the description I would expect it still to be possible to determine loadable controllers at a certain point of time (as we can only load controllers we know the configuration of)

(Btw. does this mean: https://github.com/ros-controls/ros2_control/issues/982 is solved now?)

closed 2025-05-16 by github-actions[bot]