ros-controls/ros2_control #1530: Add tests for hardware components exception handling

bmagyar, 2024-05-08:
> Is it possible to test this in any other way then defining a whole bunch of new test classes?
> 
> I'm thinking about setting up a mocker that throws on every call: https://stackoverflow.com/a/61899414

@bmagyar It's a very nice idea. do you want me to do it in this PR, or can I add it in a separate PR in the upcoming days, in the meantime, we can create an issue from your comment and I can assign it to me.

_Originally posted by @saikishor in https://github.com/ros-controls/ros2_control/issues/1508#issuecomment-2094371564_

christophfroehlich, 2024-08-29:
And for #1729

SantoshGovindaraj, 2024-10-09:
Hello @saikishor, Is this issue open for non-member contributors? 
I would like to contribute to this PR.

saikishor, 2024-10-09:
Hello @SantoshGovindaraj 

If you are confident to handle this, go ahead. Try to do it using a mocker, so we don't have to create multiple test classes.

Thank you for taking care of this :)

saikishor, 2025-11-18:
Wow! I lost this issue, and I'm sorry for not responding an eternity earlier. 

I think the easiest and better way to add tests to this is to add some parameters to the hardware components, like `throw_on_read` and `throw_on_write` etc, and then use these parameters within the existing plugins of the `hardware_interface_testing` package to use this information and throw accordingly. This way can test all the possible cases.

shlok-mehndiratta, 2026-04-19:
This issue is addressed in PR #3228 with a comprehensive verification suite and deterministic failure injection. The PR is now finalized and ready for maintainer review.