ros-controls/ros2_control #1546: Using controller with collaborative/softening mode.

LordAbraham, 2024-05-22:
Hi,

I have unique problem, I need to use KUKA iiwa in softening mode when grasping an object, system have no problem with working robot in soft mode, but when I'm switching this mode off, robot jumps to last position that was commanded by moveit.
The only solution that I found is to use moveit API to command going to current position, then cutting out trajectory between last commanded pose and current pose.

I wonder if there is a proper way to set target position for ros2 control, or reset it to current position, parallel to moveit.
For better picture, this feature would be analogic to using a robot in collaborative mode, for example:
We're setting robot in collaborative mode, then we're moving it to different position by hand, and then we're turning off collaborative mode.
Now, in that case, the robot would try to run to the last commanded position with full speed.

My controller is based on position controller from this [repo](https://github.com/ICube-Robotics/iiwa_ros2).

Thanks in advance,
Abraham

christophfroehlich, 2024-05-22:
Hi,
how do you enable the softening mode? Via ROS, or by some buttons on the robot itself? 
Is the position controller active during the collaborative mode? Why don't you set it inactive, do your manual repositioning, reactivate the controller and take the current position as new goal at the time of reactivation?

LordAbraham, 2024-05-23:
Many thanks for reply,
Old school, turning it on and off done the trick. Now, when sending a command to a robot to turn off softening, I quickly deactivate and activate the controller.
Before I was looking for a way of reseting position to current one, but if it works I'll stick to it for now.
@christophfroehlich Thank you for your help.

closed 2024-05-23 by LordAbraham