ros-controls/ros2_control #1578: Is it possible to implement event-driven control within the ROS 2 Control framework?

lkx8421, 2024-06-18:
I am currently facing a problem with my feedback control system: the feedback sensor data is obtained through subscribing to topics and is assigned to the hardware interface. When I launch the control, there is a slight delay between the actual control computation start and the sensor timestamp (the control frequency is 500 Hz, which means the control computation may start 0-2ms later than the sensor timestamp). Here is the topic timestamp and control start timestamp.
[image]

[image]

Have you ever encountered a similar problem? If so, how did you solve it? Looking forward to your reply!

christophfroehlich, 2024-06-19:
Can you give more details about the signal paths? You have written a custom hardware component, which is subscribing to topics an publish them again?

lkx8421, 2024-06-20:
Thank you for your reply. Indeed, I assigned values to the hardware component by subscribing to the motors information. These signals are for displaying the information of the three groups of sensors I subscribed to and the timestamp at the start of control (the blue straight line). I can't specify that the control starts to be executed immediately after the three topics are actually received,  it is executed after an uncertain period of delay.

closed 2025-05-16 by github-actions[bot]