ros-controls/ros2_control #1598: How to set/unset a controller to chained mode?

francescodonofrio, 2024-07-01:
Hello to everyone, I'd like to know if it has been implemented a way to set/unset a chainable controller state to chained mode, for example via a service.

christophfroehlich, 2024-07-15:
Hi! What do you want to do? If you controllers in a chain are activated/deactivated, then the chained mode will automatically set by the controller_manager.
https://github.com/ros-controls/ros2_control/blob/0c17054773d2962f695cdc3b0650aa20a3ba9c92/controller_interface/include/controller_interface/controller_interface_base.hpp#L236-L256

francescodonofrio, 2024-07-24:
Hi @christophfroehlich, my initial idea was to allow the user to dynamically reconfigure the controller to chained or unchained mode, and this was the reason of my question. Analyzing the implementation of the set_chained_mode method in [ChainableControllerInterface](https://github.com/ros-controls/ros2_control/blob/0c17054773d2962f695cdc3b0650aa20a3ba9c92/controller_interface/src/chainable_controller_interface.cpp) I've seen that this is not possible, because the controller has to be in UNCONFIGURED state in order to allow to set it to chained mode.
By the way into the implemenation of some controllers that implements the ChainableControllerInterface, like for example the admittance controller, it is not clear to me how the chained mode should be set, given that the function mentioned by you is never used.

christophfroehlich, 2024-07-24:
> By the way into the implemenation of some controllers that implements the ChainableControllerInterface, like for example the admittance controller, it is not clear to me how the chained mode should be set, given that the function mentioned by you is never used.

This is not called from the controllers but from the controller_manager
https://github.com/ros-controls/ros2_control/blob/3bc4e456dd4846a277b8f898183d30d0c9f56680/controller_manager/src/controller_manager.cpp#L1513

francescodonofrio, 2024-07-24:
@christophfroehlich thank you, I misunderstood that.
How is it possible to indicate the controller manager that a controller has to be switched to chained mode?

francescodonofrio, 2024-08-16:
@christophfroehlich after analyzing the code of the controller manager and giving a better reading to the [controller chaining documenation](https://control.ros.org/rolling/doc/ros2_control/controller_manager/doc/controller_chaining.html), I've understood that a chainable controller commutes to chained mode when its reference interfaces are claimed by another controller.

In particular, when the controller manager activates a controller it verifies the required command interfaces, and if these correspond to the reference interfaces of a certain controller, such controller is commuted to chained mode.

Thank you for your help!

closed 2024-08-16 by francescodonofrio

christophfroehlich, 2024-08-16:
Is there anything missing in the documentation to make that clear? If you have suggestions to improve that, this is very welcome!

francescodonofrio, 2024-08-16:
> Is there anything missing in the documentation to make that clear? If you have suggestions to improve that, this is very welcome!

I think that the documentation is in line with the rest of the ros's documentation.

By the way it would be a good thing to give some extra informations about how the controller state is changed, or even to share a link to the section of the controller manager in which this is done, for those interested.

Thank you for your work!